zephyrproject-rtos/zephyr #21514: Logging - strange behaviour with RTT on nRF53

koffes, 2019-12-19:
Currently using the logger with mostly default settings on the nRF53.
Whilst using LOG_WRN in a high frequency loop, I would expect to see something like: 
`--- 71 messages dropped ---`
This happens and is OK.

If I set CONFIG_LOG_BLOCK_IN_THREAD=y the same thing happens. I would expect this to block the thread and output all messages from the for-loop.

If CONFIG_LOG_IMMEDIATE=y is set, I would expect to get all logs, but there is a gap in the prints from the for loop, and no error message.

@nordic-krch 

**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

nordic-krch, 2019-12-19:
@koffes will take a look.

closed 2019-12-19 by nashif

nordic-krch, 2020-01-02:
> If CONFIG_LOG_IMMEDIATE=y is set, I would expect to get all logs, but there is a gap in the prints from the for loop, and no error message.

I've been looking into that a bit and i think that it is host related. When there is a debugging session opened (e.g. Ozone) and RTT viewer then actually both are trying to read RTT data. Most is read by RTT viewer but some is read by debug session.

I've been successfully reading all logs (on nrf52 and nrf53) when i was running Ozone debug session and use PuTTY to connect to telnet 127.0.0.1:19021.

However, I've found one issue that was causing RTT performance degradation in immediate mode (#21644).

koffes, 2020-01-02:
Interesting. AFAIK, the debug session should read from a different channel than RTT. I am quite certain that I did not have a debug session running whilst reading RTT.  Is this something that can be described and sent to Segger?

nordic-krch, 2020-01-02:
Without debug session, RTT viewer only i see some occasional bytes loss (1 per 1000 entries). I don't see them when using Ozone+PuTTY. When I use Ozone+RTT viewer i see multiple losses on RTT viewer (that's competing for RTT data that i've described above).

So it seems that there are 2 issues that could be reported to segger:
- debug session + rtt viewer
- occasional losses on rtt viewer

XavierChapron, 2020-04-23:
As @koffes , I am still having issues with #21517
I have created a PR that resolves it (at least in my situation): https://github.com/zephyrproject-rtos/zephyr/pull/24647
Please have a look.

PR #21517 (2019-12-19): logging: Fix synchronous logging in thread context
Due to flipped logic is_irq_locked function was returning true then
interrupts were unlocked. Because of that CONFIG_LOG_BLOCK_IN_THREAD
feature was not working correctly and wasn't locking thread context
when log message buffer pool was empty.

Fixes #21514.