zephyrproject-rtos/zephyr #21532: can not build the image ,No targets specified and no makefile found

lemin9538, 2019-12-20:
I follow the guide https://zephyr-doc.readthedocs.io/zh_CN/latest/getting_started/getting_started.html#id4 to start the compile, but I always get this error

" No targets specified and no makefile found "

for a new developer, I think the build system of zephyr is too complex, not very firendly for developers:
1 - need install many  tools, some of the tools are blocked in china or with slow download speed, I spend almost 2 days to finish the evn setup.

lemin9538, 2019-12-20:
seems I am using the wrong docments, I follow below documents works ok

https://docs.zephyrproject.org/latest/getting_started/index.html

mbolivar, 2019-12-20:
Thanks for following up. I'm closing this since you reported the correct documentation is working for you.

closed 2019-12-20 by mbolivar

dbkinder, 2019-12-20:
Oh dear, I don’t know about the docs at https://zephyr-doc.readthedocs.io/zh_CN/latest/index.html but notice they’re for the v1.7 branch and quite old. I’ll try to find out who’s maintaining them and if they should be deleted.

lemin9538, 2019-12-23:
Yes, I see this doc marked as latested version, so I think this is the right steps to build zephyr. 

Thanks for your support.