zephyrproject-rtos/zephyr #21536: tests/subsys/fs/fat_fs_api fails on native_posix_64

aescolar, 2019-12-20:
**Describe the bug**
The testcase tests/subsys/fs/fat_fs_api fails on native_posix_64

**To Reproduce**
1. mkdir build; cd build
2. cmake -DBOARD=native_posix_64 ../tests/subsys/fs/fat_fs_api/ -GNinja
3. ninja
4. zephyr/zephyr.exe

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Broken master in CI

**Screenshots or console output**
```
*** Booting Zephyr OS build zephyr-v2.0.0-2832-g2a5d61d3ea0d  ***
Running test suite fat_fs_basic_test
===================================================================
starting test - test_fat_mount
ASSERTION FAIL [(res == FR_OK)] @ 
subsys/fs/fat_fs.c:353
	FS init failed (-5)
@ lib/os/assert.c:39:
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x0000557e2685a980 (unknown)
E: Halting system
Exiting due to fatal error
```

**Environment (please complete the following information):**
 - OS: Unbuntu 18.04
 - Toolchain: host gcc
 - Commit SHA: Today's master 2a5d61d3ea0de47250f1a04bc8ae72fe275fc18e

aescolar, 2019-12-20:
For whoever wants to look into this, it is a un-initialized memory fest. Here the first few errors of many more
```
starting test - test_fat_mount
==10197== Thread 2:
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115D21: f_mkfs (ff.c:5247)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
==10197== 
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115D26: f_mkfs (ff.c:5247)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
==10197== 
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115DB3: f_mkfs (ff.c:5276)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
==10197== 
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115DC1: f_mkfs (ff.c:5279)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
==10197== 
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115F74: f_mkfs (ff.c:5477)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
==10197== 
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115F9E: f_mkfs (ff.c:5480)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
==10197== 
==10197== Conditional jump or move depends on uninitialised value(s)
==10197==    at 0x115EB7: f_mkfs (ff.c:5502)
==10197==    by 0x112BEB: fatfs_mount (fat_fs.c:345)
==10197==    by 0x1128CD: fs_mount (fs.c:548)
==10197==    by 0x10E7E2: test_mount (test_fat_mount.c:30)
==10197==    by 0x10E7E2: test_fat_mount (test_fat_mount.c:41)
==10197==    by 0x1130F9: run_test_functions (ztest.c:148)
==10197==    by 0x1130F9: test_cb (ztest.c:281)
==10197==    by 0x10ED9C: z_thread_entry (thread_entry.c:29)
==10197==    by 0x10FE86: posix_thread_starter (posix_core.c:305)
==10197==    by 0x51E16DA: start_thread (pthread_create.c:463)
==10197==    by 0x551A88E: clone (clone.S:95)
```

pabigot, 2019-12-21:
The underlying cause is probably described in #19231.

PR #21668 (2020-01-03): tests filesystem.fat.api: Do not run on native_posix_64
The FAT FS code clearly does not support platforms where
long is 64bits. See
https://github.com/zephyrproject-rtos/fatfs/blob/master/include/integer.h#L30

So, do not try to run it in native_posix_64
Fixes #21536
Fixes #19231
commit 5d94c989 (2020-01-03): tests filesystem.fat.api: Do not run on native_posix_64

The FAT FS code clearly does not support platforms where
long is 64bits. See
https://github.com/zephyrproject-rtos/fatfs/blob/master/include/integer.h#L30

So, do not try to run it in native_posix_64
Fixes #21536
Fixes #19231
closed 2020-01-03 by nashif