zephyrproject-rtos/zephyr #21545: Add support for CBOR Object Signing and Encryption (COSE)

motet, 2019-12-20:
## Introduction

There are a number of new standards for IoT security under development at IETF which depend on CBOR Object Signing and Encryption, or COSE (see [IETF RFC 8152](https://tools.ietf.org/html/rfc8152)). This is the CBOR equivalent of JSON Web Encryption (see [IETF RFC 7516](https://tools.ietf.org/html/rfc7516)). Adding a COSE library is a necessary prerequisite before the following new or proposed IETF standards can be implemented in Zephyr:

- Object Security for Constrained RESTful Environments, or OSCORE (see [IETF RFC 8613](https://tools.ietf.org/html/rfc8613)), a protocol for application-layer encryption of CoAP messages.
- Authentication and Authorization for Constrained Environments, or SUIT (see [IETF working group](https://datatracker.ietf.org/wg/ace/about/)), the IoT equivalent of OAuth 2.0. This architecture enables token-based access control using CBOR Web Token, or CWT (see [IETF RFC 8392](https://tools.ietf.org/html/rfc8392)).
- Software Updates for the Internet of Things, or SUIT (see [IETF working group](https://datatracker.ietf.org/wg/suit/about/)), a firmware update architecture for IoT devices.
- Remote ATtestation ProcedureS, or RATS (see [IETF working group](https://datatracker.ietf.org/wg/rats/about/)), a remote attestation architecture for IoT devices.
- Trusted Execution Environment Provisioning, or TEEP (see [IETF working group](https://datatracker.ietf.org/wg/teep/about/)), a protocol for interoperability between TEEs running on different architectures.

There is a growing interest in the above protocols from security researchers and certification bodies (at least in the EU), and it would be great to see this key building block available in Zephyr so others can get involved on testing implementations.

## Detailed RFC

There are few existing implementations of COSE, and none are built on the encryption and CBOR libraries supported in Zephyr (i.e., TinyCrypt and tinyCBOR). It would appear that the most expedient approach is to start a new implementation from scratch.

### Proposed change (Detailed)

All of the prerequisite building blocks for COSE should exist in TinyCrypt and tinyCBOR. It seems appropriate to add the new source code in the same directories as the existing JSON source code, given the number of parallels between the two:

- `lib/os/cose.c`
- `include/data/cose.h`
- `include/data/cwt.h`
- `tests/lib/cose`

### Dependencies

All of the prerequisite building blocks for COSE should exist in TinyCrypt and tinyCBOR.

### Concerns and Unresolved Questions

My work in the near future will depend on a working COSE implementation in Zephyr. I am new to this community I'd appreciate any feedback, and to hear from anyone interested before I get started working on this. (In particular, whether the file structure described above is appropriate for this type of feature addition.)

motet, 2020-01-14:
I have a partial COSE implementation available as an external [Zephyr module](https://github.com/motet/cozy) for anyone interested.

jjmilburn, 2020-03-31:
Interesting! I'll be watching this for `MAC/MAC0` support if that gets rolled in.

carlescufi, 2021-01-13:
> I have a partial COSE implementation available as an external [Zephyr module](https://github.com/motet/cozy) for anyone interested.

@lindemer would you be willing to submit this to Zephyr as a Pull Request? We are interested in COSE and would likely review and contribute.

carlescufi, 2021-01-13:
@rlubos just found this:
https://github.com/Fraunhofer-AISEC/c_OSCORE
@oberien would you be interested in contributing OSCORE to Zephyr?

lindemer, 2021-01-14:
@carlescufi Are you mainly interested in OSCORE or do you need a generic COSE implementation? If it's the latter case I can work on this a bit and submit it. My COSE implementation depends on an [LGPL 2.1-licensed CBOR library](https://github.com/bergzand/NanoCBOR/tree/59bbe085f7f332e0834ad4857ff307fd2c9d498e) so I could use some clarification on whether that's compatible with Apache 2.0.

StefanHri, 2021-01-14:
@carlescufi  and @oberien 
c_OSCORE was developed at Fraunhofer AISEC as a Student project. Meanwhile, we further developed and extended it. The new version is still not on GitHub but we plan to do that in the next weeks. 

Are you also interested in [EDHOC](https://datatracker.ietf.org/doc/draft-selander-ace-cose-ecdhe/)? EDHOC can be used to establish symmetric keys for OSCORE. We have also an EDHOC implementation that together with OSCORE forms a light alternative to DTLS.

We will be happy to contribute both OSCORE and EDHOC to ZephyrOS. Are you interested?

carlescufi, 2021-01-15:
> @carlescufi Are you mainly interested in OSCORE or do you need a generic COSE implementation? If it's the latter case I can work on this a bit and submit it. My COSE implementation depends on an [...]

@lindemer We're mostly interested in OSCORE, but having a COSE that is not bound to it would be preferable. Unfortunately we cannot accept any copyleft licenses (including LGPL) in the main Zephyr tree, so using that CBOR library is not an option for this work to be included in the main upstream repository. Let us also discuss with @StefanHri and see where we can go from here.

carlescufi, 2021-01-15:
> @carlescufi and @oberien
> c_OSCORE was developed at Fraunhofer AISEC as a Student project. Meanwhile, we further developed and extended it. The new version is still not on GitHub but we plan to do that in the next weeks.

Great to hear, thanks for the info

> Are you also interested in [EDHOC](https://datatracker.ietf.org/doc/draft-selander-ace-cose-ecdhe/)? EDHOC can be used to establish symmetric keys for OSCORE. We have also an EDHOC implementation that together with OSCORE forms a light alternative to DTLS.

Yes, we are indeed interested in EDHOC as well.

> We will be happy to contribute both OSCORE and EDHOC to ZephyrOS. Are you interested?

Absolutely. Assuming your code uses a compatible license, I think this should go in the main tree, and not as a module. Ideally we´d need it to be licensed under the same Apache v2 license we use for the rest. Here is what I can offer to help out in the effort:

- Extensive code review and suggestions once the Pull Request(s) is posted
- Comprehensive testing on our (nRF) platforms
- Technical help to expedite resolving the issues raised in the Pull Request(s)

nashif, 2026-10-30:
OSCORE/EDHOC (the main COSE consumers) available via the uoscore-uedhoc module; no generic COSE library effort since 2021.

closed 2026-10-30 by nashif