zephyrproject-rtos/zephyr #21546: SPI broken for STM32L1

Mani-Sadhasivam, 2019-12-20:
**Describe the bug**
With the current Zephyr master, SPI is not working as expected on STM32L1 based 96boards Wistrio board. Some of the bytes of the received data seems to be dropping.

**To Reproduce**
Reproducible with the LoRa PR #18998 which uses SPI for communicating to the Radio.

The issue seems to be with this commit: https://github.com/zephyrproject-rtos/zephyr/commit/2ce8fa1e42d3d60e4119f15e82c29d2daaf67c00
To be more precise, setting `xfer_16` flag manually to `false` fixes the issue, which means the problem is with sending/receiving 16bits of data.

Mani-Sadhasivam, 2019-12-20:
@psidhu Since you are the actual author of the commit, could you please look into this?

erwango, 2019-12-22:
@Mani-Sadhasivam, thanks for reporting. This is partly my bad, not enough tests have been made for this behavior to be made used by default.
@psidhu, can you push a change to set this behavior available only under specific Kconfig flag (default n)?

psidhu, 2020-01-02:
@erwango Sure, though it may take me a small bit of time to get to.

PR #21722 (2020-01-07): Revert "spi: stm32: Enable sending 16bits at a time in 8bit mode if p…
…ossible"

This reverts commit 2ce8fa1e42d3d60e4119f15e82c29d2daaf67c00.

Fixes #21546 
Fixes #21679

Following merge of https://github.com/zephyrproject-rtos/zephyr/pull/20948, following errors have been reported:
* https://github.com/zephyrproject-rtos/zephyr/issues/21546
* https://github.com/zephyrproject-rtos/zephyr/issues/21679

We've tried several changes and fixes on the change (cf #21712)
- Set the optimized behavior optional (default n)
- 3 additional fixes on the shift frame function flow 

Though issues remain in [... cut]
commit ca61847a (2020-01-07): Revert "spi: stm32: Enable sending 16bits at a time in 8bit mode if possible"

This reverts commit 2ce8fa1e42d3d60e4119f15e82c29d2daaf67c00.

Fixes #21546
Fixes #21679
closed 2020-01-07 by jhedberg