zephyrproject-rtos/zephyr #21549: i2c_sam0 interrupt latency is excessive

sslupsky, 2019-12-20:
**Describe the bug**
I've been debugging some driver issues with the i2c_sam0 driver.  Along the way I have observed the interrupt latency for an i2c transaction is on the order of 360us.  That is, each byte of an i2c transfer takes 360us to complete which is 300% - 1000% longer than the actual duration of the i2c byte frame which is about 25us at 400 KHz or 95us at 100 KHz.  I have attached a screen shot taken with an i2c clock frequency of 100 KHz.

I have confirmed that i2c_sam0_isr() is firing every 360us during the i2c_transfer() and there is a long silent period between the byte transfers where the i2c clock remains low.  I am contemplating that the i2c interrupt that occurs at the end of the data transfer is being deferred by the kernel for some reason.  I do not fully understand Zephyrs scheduling so I want to ask someone to explain what happens with interrupts during k_sem_take()?  I read somewhere that all interrupts run at a lower priority than the kernel.  So, if the kernel is waiting for a semaphore, does that mean that interrupts are not processed until the next kernel "tick"?

I should mention that I have tried using the IRQ_DIRECT_CONNECT macro instead but this made no difference to the latency.  Also, I am using the 32khz oscillator as the source for the rtc counter.  I noticed a threshold of 7 ticks in z_clock_set_timeout() which is about 214us and is close to the amount of time between byte frames.

[image]

nordic-krch, 2020-01-08:
it might be something atmel specific. I've tried it on nrf52 and see no such delays. I tried with device power management on and off.

sslupsky, 2020-01-08:
If I enable the DMA channel, I can eliminate all of the intra frame bus idling except for the last byte.  The last byte received during a message read is not read by the DMAC but is triggered by an interrupt.  The stop bit at the end of a message write is similarly triggered by an interrupt.  In bother cases, there is usually about 360us of delay.  That is not always the case though.  Sometimes the delay is longer and sometimes it can happen immediately.

sslupsky, 2020-01-08:
It is not preferable to the the dmac though.  Using the dmac has the unfortunate side effect of consuming about 1.5K of RAM.

sslupsky, 2020-01-08:
@nordic-krch At the moment that is what I think as well.  Something appears to be blocking the interrupt for a period of time some reason but I haven’t been able to determine the sequence of events that leads to this.  The problem maybe correlated to sleep modes.

There are some very odd peculiarities with ATMEL peripherals.  In particular, reading or writing a “core” peripheral register requires a “sync”.  This is particularly problematic with regards to the rtc.  In order to read the rtc count register, this “sync” must be used and can take 183us if the rtc clock is 32khz.  The way the kernel implements setting timeouts in tickless mode, this rtc block causes significant delay because the count register is read multiple times.  I think this is likely the underlying root cause of the problem I am encountering though I am not certain about that yet.

sslupsky, 2020-01-13:
I believe I have narrowed down the cause to the scheduler.  The driver waits on a semaphore between messages of the transaction.  When it does so, the kernel can end up inserting a delay anywhere from a few 10's of microseconds to 100's of milliseconds.  If I busy wait for the semaphore, I can eliminate the delay.

This suggests perhaps an issue exists with setting timeouts in the kernel.  I read an issue that was posted a while back regarding timeout issues with the scheduler.  Perhaps this maybe related.  Something is delaying the context switch back to the waiting thread.

closed 2020-08-28 by github-actions[bot]