zephyrproject-rtos/zephyr #21551: GPIO Interrupt is not triggered for range GPIO240_276

albertofloyd, 2019-12-21:
No interrupt is triggered when select GPIO in range  GPIO240_276, interrupt is not triggered

Modify gpio_api test and select  MCHP_GPIO_254 

Callbacks should occurred whenever GPIO 254 changes level

franciscomunoz, 2020-01-06:
GPIO240,276 isrs are failling because GIRQ22 is missing from the enum 

312 enum MCHP_GIRQ_IDS {
313         MCHP_GIRQ08_ID = 8,
314         MCHP_GIRQ09_ID,
315         MCHP_GIRQ10_ID,
316         MCHP_GIRQ11_ID,
317         MCHP_GIRQ12_ID,
318         MCHP_GIRQ13_ID,
319         MCHP_GIRQ14_ID,
320         MCHP_GIRQ15_ID,
321         MCHP_GIRQ16_ID,
322         MCHP_GIRQ17_ID,
323         MCHP_GIRQ18_ID,
324         MCHP_GIRQ19_ID,
325         MCHP_GIRQ20_ID,
326         MCHP_GIRQ21_ID,
327
328         MCHP_GIRQ23_ID,
329         MCHP_GIRQ24_ID,
330         MCHP_GIRQ25_ID,
331         MCHP_GIRQ26_ID,
332         MCHP_GIRQ_ID_MAX,
333 };

located in the HAL, file : microchip/mec/mec1501/component/ecia.h.

@scottwcpg : Would you like to update the HAL ?

franciscomunoz, 2020-02-13:
This issue was already solved with  a HAL change. Therefore, I'm closing it.

closed 2020-02-13 by franciscomunoz