zephyrproject-rtos/zephyr #21552: Constant disconnects while attempting BT LE multi-central application.

mpdas108, 2019-12-21:
I am attempting a multi-central application, using zephyr-v2.1.0-335-gce753ea49113, and a Nordic nrf52840. I am able to connect to the centrals, and discover and subscribe to a characteristic, but they all disconnect after about 10 seconds with BT_HCI_ERR_CONN_TIMEOUT.  They ten proceeded to reconnect without a problem, but then disconnect again, and repeat like that. 

Any suggestions? I can provide what ever other information is needed to help debug this. I have attached a copy of my program.

[zephyr-multi-central.zip](https://github.com/zephyrproject-rtos/zephyr/files/3990984/zephyr-multi-central.zip)

aescolar, 2019-12-23:
CC @cvinayak @carlescufi

cvinayak, 2019-12-24:
Some observations in your application:
1. Explicitly selected, CONFIG_CLOCK_CONTROL_NRF_K32SRC_XTAL=y, but using default 20ppm accuracy, would you want to set the value for the XTAL used in your board? Incorrrect accuracy can lead to connection failures.
2. Support for 32 simultaneous connections, under worst case BLE connection events would temporally overlap, effectively each connection skipping 31 of consecutive connection events. The XTAL accuracy needs to be accurate for the peripherals to open up Rx window correctly. Also, supervision timeout values needs to consider these skipped events.
3. Scanning is restarted on new connection establishment, to discover and connect to new devices, the scan window of 50ms is used. The connection interval is 7.5 ms, which means 6 connection events can be skipped due to the 50 ms window, this would be observed as disconnection with reason 0x3E.
4. You are using PA/LNA and also nRF Radio at +4dBm Tx power, which PA/LNA front end are you using? and could you check if the issue exists with nRF Radio at 0dBm Tx power.
5. Controller is configured for 251 byte PDU, but I dont see corresponding increase considered for Host buffers.

Action:
1. Try to set higher xtal accuracy value (or the one spec-ed by the xtal manufacturer)
2. Use large connection interval and corresponding supervision timeout, considering the overall number of overlapping connections per interval.
3. Use smaller scan window size than the connection interval, to allow connection established to succeed (with in 6 connection events).
4. Check if the problem is solved without the use of PA/LNA.

mpdas108, 2019-12-24:
Hello cvinayak,

Thank you for your feedback. This application is far from complete, I have a lot more work to do. I am just trying to get things going, and I am running into these issues. I have a complete working firmware for our previous version of this board using the Nordic SDK 14.2.0 and the Fanstel BT832XE. We successfully connect up to 20 devices with no issues. We wanted to add some features and upgrade the module to the Fanstel BT840XE, and hopefully connect to more devices. The BT840XE requires the Nordic SDK 15.3.0 and have found limitations and issues with the new Nordic SDK. So to facilitate connecting to more devices and hopefully a more stable environment we are exploring using Zephyr as the base for the new firmware.

In regard to you suggested actions:

1. I am using a Fanstel BT840XE module on our board, this module has a LF XTAL on the module. I have not been able to find anything in their documentation on the specific accuracy of the module. I have even taken the module apart to try and discover the specific XTAL they use to no avail, as the markings on the XTAL are generic. I then went and tried every possible accuracy setting just to see if a different accuracy would produce better results, all produce the same result. So I left it on the default accuracy. Is there an accuracy setting higher than 20ppm configurable?
2. and 3. This sounds promising, I will try adjusting these settings and get back to you. As of right now I only have 12 devices I am trying to connect to, but I will need to eventually connect to more. Therefore I have it set to the number I eventually want to connect to. If I find out that It is impossible I still want to try and discover what is the actual maximum I can connect to.
4. The Fanstel BT840XE module has a SKY66112 amplifier. I have also tried with Tx power set to 0dBm and the PA/LNA disabled, with the same results.

As for the 251 byte PDU, I will eventually need to use that, but as of now I am not using that size. I have also reduced the size back to defaults and I get the same results.

I will try adjusting scan window, connection interval, and supervision timeout in a few days when holidays are over, and let you know how it went. Although these are the same amounts that I am using in our previous firmware written with the Nordic SDK in which I can successfully stay connected to 20 devices.

cvinayak, 2019-12-26:
I suggest you give a try with CONFIG_BT_LL_SW_LEGACY=y too and let me know the results.

mpdas108, 2019-12-26:
Thanks, I will let you know how it goes. As far as accuracy of an XTAL, the lower ppm number means higher/better accuracy, 20ppm is higher accuracy than 500ppm. That would mean a value 10ppm is higher accuracy than 20ppm.

mpdas108, 2019-12-30:
So yeah I have tried all of your suggestions, and fixed a couple of other issues I had. But they still disconnect.

mpdas108, 2019-12-31:
I also reduced the devices I try to connect to down to 3, and they still disconnect

mpdas108, 2020-01-03:
For anyone else that finds this thread. I have given up, and we are shifting to using ANT protocol and the Nordic SDK to connect many sensors.

aescolar, 2020-02-14:
Closing as the discussion seems to have reached the end. If you would like to continue please reopen.

closed 2020-02-14 by aescolar