zephyrproject-rtos/zephyr #21554: ldscript:datas section is not properly aligned in ROM

LuoZhongYao, 2019-12-21:
If the noinit section is not exactly aligned, the datas section will not be aligned in ROM.

**Describe the bug**
```
$ git diff
diff --git a/samples/basic/blinky/src/main.c b/samples/basic/blinky/src/main.c
index d5e99bbd7e..80601e0ce8 100644
--- a/samples/basic/blinky/src/main.c
+++ b/samples/basic/blinky/src/main.c
@@ -14,11 +14,13 @@
 /* 1000 msec = 1 sec */
 #define SLEEP_TIME     1000
 
+char __noinit t = 0;
 void main(void)
 {
        u32_t cnt = 0;
        struct device *dev;
 
+       t = 1;
        dev = device_get_binding(LED_PORT);
        /* Set LED pin as output */
        gpio_pin_configure(dev, LED, GPIO_DIR_OUT);

$ export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
$ export ZEPHYR_SDK_INSTALL_DIR=$ZEPHYR/zephyr
$ source zephyr-env.sh
$ west build -p auto -b stm32_min_dev_blue samples/basic/blinky
$ readelf -l build/zephyr/zephyr.elf

Elf file type is EXEC (Executable file)
Entry point 0x8000be5
There are 4 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x002854 0x080027a0 0x080027a0 0x00008 0x00008 R   0x4
  LOAD           0x0000b4 0x08000000 0x08000000 0x02b4c 0x02b4c RWE 0x4
  LOAD           0x002c00 0x20000ee4 0x08002b4f 0x000ec 0x000ec RW  0x4
  LOAD           0x002cf0 0x20000000 0x20000000 0x00000 0x00ee1 RW  0x8

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx 
   01     text _TEXT_SECTION_NAME_2 .ARM.exidx sw_isr_table devconfig rodata 
   02     datas initlevel _k_mutex_area 
   03     bss noinit 

```

andrewboie, 2019-12-23:
we really need you to state what platform you are building for.
have you tried adding an alignment directive in the relevant linker script?

LuoZhongYao, 2019-12-23:
> 
> 
> we really need you to state what platform you are building for.
> have you tried adding an alignment directive in the relevant linker script?

OS: Windows 10.0.18363.535,  WSL1 + Fedora-Remix-for-WSL, 
SDK: 0.10.3
Except for adding test code in samples/basic/blinky/src/main.c, no files have been modified.

andrewboie, 2019-12-30:
> OS: Windows 10.0.18363.535, WSL1 + Fedora-Remix-for-WSL,
> SDK: 0.10.3
> Except for adding test code in samples/basic/blinky/src/main.c, no files have been modified.

That doesn't answer my question.
what platform you are building **for**
ARM? what board?

LuoZhongYao, 2019-12-30:
> > OS: Windows 10.0.18363.535, WSL1 + Fedora-Remix-for-WSL,
> > SDK: 0.10.3
> > Except for adding test code in samples/basic/blinky/src/main.c, no files have been modified.
> 
> That doesn't answer my question.
> what platform you are building **for**
> ARM? what board?

stm32_min_dev_blue

closed 2020-08-28 by github-actions[bot]