zephyrproject-rtos/zephyr #21568: mps2_an385:tests/kernel/tickless/tickless_concept/kernel.tickless.concept  fail

wentongwu, 2019-12-23:
*** Booting Zephyr OS build zephyr-v2.1.0-580-gb0a8461a873a  ***
Running test suite tickless_concept
===================================================================
starting test - test_tickless_sysclock
time 10, 220
time 230, 430
PASS - test_tickless_sysclock
===================================================================
starting test - test_tickless_slice
elapsed slice 110, expected: <100, 110>
elapsed slice 120, expected: <100, 110>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:59: thread_tslice: t <= SLICE_SIZE_LIMIT is false

elapsed slice 0, expected: <100, 110>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

elapsed slice 0, expected: <100, 110>

    Assertion failed at ZEPHYR_BASE/tests/kernel/tickless/tickless_concept/src/main.c:57: thread_tslice: t >= SLICE_SIZE is false

FAIL - test_tickless_slice
===================================================================
Test suite tickless_concept failed.
===================================================================
PROJECT EXECUTION FAILED

galak, 2020-01-07:
Is this failure on hardware or QEMU?

jhedberg, 2020-01-07:
Where did you see the issue? In CI?

wentongwu, 2020-01-08:
> Is this failure on hardware or QEMU?

QEMU

wentongwu, 2020-01-08:
> Where did you see the issue? In CI?

yes, in CI

stephanosio, 2020-01-08:
This happens when the host CPU usage is very high (e.g. when running sanitycheck).

Re-running the test should make it pass (it is an intermittent failure caused by processing delay).

stephanosio, 2020-01-08:
cc @nashif

stephanosio, 2020-01-08:
> Where did you see the issue? In CI?

I have seen this failure a number of times in the CI. Simply re-running the failing block in the Shippable will resolve this.

Maybe we should consider increasing the retry count for this test.

erwango, 2020-01-16:
@jhedberg  Can we increase the priority ? Frequency of the failure is non negligible

jhedberg, 2020-01-16:
> @jhedberg Can we increase the priority ? Frequency of the failure is non negligible

@erwango do you know who might be able to take ownership of this issue (i.e. have it assigned to them)? We can change the priority, but without an assignee I'm afraid it will not make much difference.

erwango, 2020-01-16:
> @erwango do you know who might be able to take ownership of this issue

@jhedberg, if there is not board owner then maybe the subsystem owner ?

erwango, 2020-01-20:
@karl-zh since you're familiar with mps2 boards, can you have a look to this issue (it is annoying since poping up often in shippable reports)?

andrewboie, 2020-01-21:
@wentongwu you have already been looking into the timing related issues with QEMU targets

why did you open this bug? why do you think this is unrelated to the QEMU timing issues we have been aware of for a long time now?

> @jhedberg Can we increase the priority ? Frequency of the failure is non negligible

@erwango this is not being seen on real hardware

This almost certainly an instance of #14173 unless someone has evidence to the contrary

wentongwu, 2020-01-22:
@andrewboie Yes I'm looking at it, just surprised we have already enabled icount on mps2_an385 with "-icount auto" which I don't think is correct. I will close this issue and continue #14173 if got big slot. Thanks

closed 2020-01-22 by wentongwu

erwango, 2020-01-22:
@wentongwu why closing the issue if still present ?
I meet this failure about twice a day in failling shippable reports.
That's might not be an implementation issue, maybe this test should not be run, but I don't think thre is really nothing to do here.