zephyrproject-rtos/zephyr #21571: CONFIG_BT_CENTRAL doesnot work fine with nrf51_pca10028

stephenmasih, 2019-12-23:
bt_le_scan_start() works fine with nrf52_pca10040 but the UART console does not print anything in nrf51_pca10028. The code is all OK. When I comment CONFIG_BT_CENTRAL=y it prints in the UART console, but for that purpose, I had to also comment out in bt_le_scan_start().

[image]
[image]

Please help.

aescolar, 2019-12-23:
CC @cvinayak @carlescufi

carlescufi, 2019-12-23:
@stephenmasih for the time being, and while @cvinayak has some time to look at it, can you try enabling `CONFIG_BT_LL_SW_LEGACY=y` in your `prj.conf` and see if that changes something.

cvinayak, 2019-12-24:
@stephenmasih Tested 4e135d76a3 samples/bluetooth/central_hr on nrf51_pca10028 board, found no issues with console messages on UART. Attached console log:

[minicom1.txt](https://github.com/zephyrproject-rtos/zephyr/files/3997138/minicom1.txt)

Please share commit hash details, minimal application that reproduces the issue and steps to build and reproduce.

stephenmasih, 2019-12-24:
@carlescufi @cvinayak I figured out the problem! I created a thread to start the scanning. It doesn't work in this case. But when I defined the same scanning code inside the main() it works fine, and it starts printing in the UART console.

stephenmasih, 2019-12-24:
NRF51_PCA10028 only has one UART. In my application can I use the same uart for the console as well as for checking the AT command outputs (That's only for debugging the AT commands I am getting out from the MCU)?

cvinayak, 2019-12-26:
The issues does not appear to be Bluetooth related. Until steps to reproduce Bluetooth issue is detailed here, no further action.

stephenmasih, 2019-12-27:
Yes it is not a Bluetooth related issue. I will post a new issue. Thank you for your support.

closed 2019-12-27 by stephenmasih