zephyrproject-rtos/zephyr #21581: GNU ARM Embedded link broken in Getting Started

danieleggert, 2019-12-23:
The link to GNU ARM Embedded on https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#gnu-arm-embedded should probably be updated to https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

ulfalizer, 2019-12-23:
Thanks for the report.

Fix: https://github.com/zephyrproject-rtos/zephyr/pull/21583

carlescufi, 2019-12-28:
@danieleggert as I commented in the PR, the existing link seems to work fine? https://developer.arm.com/open-source/gnu-toolchain/gnu-rm

jhedberg, 2020-01-07:
The existing link works for me too, maybe it was a temporary issue with the website. Therefore closing this issue.

closed 2020-01-07 by jhedberg