zephyrproject-rtos/zephyr #21586: Bluetooth Mesh fail to transmit messages after some time on nRF52840

aurel32, 2019-12-24:
**Describe the bug**
I am running a small Bluetooth Mesh network, with some Zephyr nodes. After some time (usually days, but see below) some of the nodes fail to transmit data with the following error:
`<err> bt_mesh_adv: Advertising failed: err -69`

This happens even if the node doesn't transmit data besides working as a relay and transmitting secure beacons. I have tried to diagnose the issue using help from slack. I have attached a log of the issue with a few additional logging messages (the issue starts at 01:07:33.047,668). The corresponding patch is attached.

The problem is that the call to `bt_le_adv_stop` fail to stop the advertising within the 10 second timeout. Once this happen the next call to `adv_send` will fail as the advertising is already enabled and sending data is never possible anymore.

**To Reproduce**
I am using a nRF52 PCA10059 board with serial console over USB. I have attached the `prj.conf` and `main.c` files, mostly inspired from existing samples. Relay is enabled with 1 retransmission with 2 interval steps (default). Secure beacons are enabled (default). Friend is compiled but disabled.

I have the feeling the issue is related to the network environment as I can reproduce the problem with a very simple node with only the mandatory models. My network is composed of the following nodes in addition to the one I am using to debug the failure:
- A Linux based node using the bluetooth-meshd daemon from Bluez. It is mostly receiving data, sending data sporadically. It is also configured as a friend, but relay is disabled.
- A Ledvance Smart+ Classic E27 Multicolor light bulb, publishing 6 bytes every 10 seconds. Relay is enabled with 1 retransmission with 9 interval steps (default).
- A Ledvance Smart+ Indoor Plug EU, publishing 3 bytes every 60 seconds + on change. Relay is enabled with 1 retransmission with 9 interval steps (default).
- A nRF52 PCA10059 based Zephyr node publishing 9 bytes every second (vendor model) + 
5 bytes every 15 seconds (sensor model). Relay is enabled with 1 retransmission with 2 interval steps (default).
- A nRF52 PCA10059 based Zephyr node publishing 5 bytes every 15 seconds (sensor model). Relay is enabled with 1 retransmission with 2 interval steps (default).

With this setup I used to encounter the problem appears between 3 to 9 days. Since I am debugging the issue, it happens more often and I can usually reproduce the problem between 1 to 5 hours. It might be due to the additional workload of logging debug messages.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.10.3
 - Commit SHA or Version used: zephyr-v2.1.0 aka b413223a66ed9e857c67ee21d1e8be4768804dd1
- Board: NRF52840_pca10059

All the mentioned files are available in the following zip [zephyr-nrf52-btmesh-tx-fail.zip](https://github.com/zephyrproject-rtos/zephyr/files/3998730/zephyr-nrf52-btmesh-tx-fail.zip)

jhedberg, 2019-12-25:
Was it so that the initial analysis was pointing at something on the controller side, or did I misread the Slack backlog?

aurel32, 2019-12-28:
@jhedberg Yes I confirm that the discussions on Slack are pointing to a controller issue. The latest test I have done show that when the issue happens, the semaphore in `ull_disable` is never taken.

joerchan, 2020-01-03:
Log working advertising disable command
```
[01:07:01.403,015] <dbg> bt_mesh_adv.adv_send: Advertising about to be stopped
...
[01:07:01.403,045] <dbg> bt_hci_core.bt_hci_cmd_send_sync: buf 0x20017010 opcode 0x200a len 4
...
[01:07:01.403,076] <dbg> bt_hci_core.send_cmd: Sending command 0x200a (buf 0x20017010) to driver
[01:07:01.403,106] <dbg> bt_hci_core.bt_send: buf 0x20017010 len 4 type 0
[01:07:01.403,106] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
...
[01:07:01.403,137] <dbg> bt_hci_core.bt_buf_get_cmd_complete: sent_cmd 0x20017010
...
[01:07:01.403,198] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[01:07:01.403,198] <dbg> bt_hci_core.bt_hci_cmd_send_sync: opcode 0x200a status 0x00
[01:07:01.403,228] <dbg> bt_mesh_adv.adv_send: Advertising stopped
```

Log failing advertising disable command
```
[01:07:33.047,668] <dbg> bt_mesh_adv.adv_send: Advertising about to be stopped
...
[01:07:33.047,698] <dbg> bt_hci_core.bt_hci_cmd_send_sync: buf 0x20017010 opcode 0x200a len 4
...
[01:07:33.047,760] <dbg> bt_hci_core.send_cmd: Sending command 0x200a (buf 0x20017010) to driver
[01:07:33.047,760] <dbg> bt_hci_core.bt_send: buf 0x20017010 len 4 type 0
[01:07:33.047,790] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
...
[01:07:43.047,790] <dbg> bt_hci_core.bt_hci_cmd_send_sync: opcode 0x200a status 0x00
[01:07:43.047,790] <dbg> bt_mesh_adv.adv_send: Advertising stopped
```

From the output it looks like hci_cmd_send_sync has timed out after 10 seconds waiting for the command complete event.
It appears to stall before retreiving the command complete event buffer since this print statement is no longer present:
`[01:07:01.403,137] <dbg> bt_hci_core.bt_buf_get_cmd_complete: sent_cmd 0x20017010`

This means that the `le_set_adv_enable` in `controller/hci/hci.c` is where it is stuck:
```
	u8_t status;

	status = ll_adv_enable(cmd->enable);

	ccst = hci_cmd_complete(evt, sizeof(*ccst));
```

Between hci_driver_send and ll_adv_enable there are no scheduling points which is the Enter print statement.
Between hci_cmd_complete and bt_buf_get_evt there are no scheduling points, which is the bt_buf_get_cmd_complete print statement.

My suspicion was this:
```
static inline u8_t disable(u16_t handle)
{
	/* ... */
	ret = ull_disable(&adv->lll);
	LL_ASSERT(!ret);

	/* ... */
}

int ull_disable(void *lll)
{
	/* ... */
	return k_sem_take(&sem, K_FOREVER);
}
```

Which @aurel32 is confirming (correct me if I'm wrong) is the problem.

aurel32, 2020-01-08:
Yes, I confirm. I have added `LOG_DBG` in `ull_disable()` before and after taking the semaphore. When the problem happens, only the log entry before the `k_sem_take()` appears. So it's indeed stuck there.

carlescufi, 2020-01-09:
@cvinayak can you please investigate the root cause here. It seems from @joerchan 's investigation that the `mayfly_enqueue` call before the `k_sem_take` is not triggering a `k_sem_give`.

PR #21824 (2020-01-16): Bluetooth: controller: split: Fix ull_disable hang
Under race conditions it is possible that there is no call
to k_sem_give to the waiting k_sem_take in the ull_disable
function.

ull_disable function checks for reference count before
using a mayfly to schedule lll_disable, which in turn
would close requested currently active role event leading
to done event being propogated to ULL. Done event would
then call the set disabled_cb callback when the reference
count is zero, giving the semaphore to the waiting
k_sem_give in the ull_disable.

Under race conditions if the reference count reached zer [... cut]
commit b5c63c69 (2020-01-16): Bluetooth: controller: split: Fix ull_disable hang

Under race conditions it is possible that there is no call
to k_sem_give to the waiting k_sem_take in the ull_disable
function.

ull_disable function checks for reference count before
using a mayfly to schedule lll_disable, which in turn
would close requested currently active role event leading
to done event being propogated to ULL. Done event would
then call the set disabled_cb callback when the reference
count is zero, giving the semaphore to the waiting
k_sem_give in the ull_disable.

Under race conditions if the reference count reached ze [... cut]
closed 2020-01-16 by aescolar