zephyrproject-rtos/zephyr #21591: Timeout error for the Microchip board during Sanitycheck

maksimmasalski, 2019-12-26:
**Describe the bug**
Discovered problem that Sanitycheck stopped work on the Microchip board. Now for every test after it was build and uploaded to the board, no output during Sanitycheck run, as a result Sanitycheck shows Timeout error. Tested on 2 independent computers with Microchip board.

The latest working commit was 23bdddea0. Below is shown output which was during that commit.  If to use same commit now, error happens and I can't reproduce the same working behaviour. It means that something happened maybe with west. I can't find out the root cause, I tried to modify sanitycheck script, but it was no succes to deal with it.

**====WORKING OUTPUT START====**

>  [maksim@maksim-nuc zephyr]$ sanitycheck --device-testing --device-serial /dev/ttyUSB1 -p mec15xxevb_assy6853 -T /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api/ -v -v
JOBS: 8
Reading test case configuration files under /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api...
scanning /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api
Found possible test case in /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api
Reading platform configuration files under /home/maksim/zephyrproject/zephyr/boards...
platform filter: ['mec15xxevb_assy6853']
    arch_filter: None
     tag_filter: None
    exclude_tag: None
Building initial testcase list...
2 test configurations selected, 0 configurations discarded due to filters.
Device testing on:
mec15xxevb_assy6853 (None) on /dev/ttyUSB1
Adding tasks to the queue...
Running cmake on /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api for mec15xxevb_assy6853
Running cmake on /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api for mec15xxevb_assy6853
DONE FEEDING
build test: mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo
Building /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api for mec15xxevb_assy6853
build test: mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo.poll
Building /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api for mec15xxevb_assy6853
run test: mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo
make: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
run test: mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo.poll
make[1]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[2]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[  1%] Built target kobj_types_h_target
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[  4%] Built target syscall_list_h_target
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[  5%] Built target syscall_macros_h_target
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[  5%] Built target driver_validation_h_target
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[  6%] Built target offsets
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[  7%] Built target offsets_h
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[ 24%] Built target kernel
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[ 28%] Built target app
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[ 46%] Built target zephyr
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[ 47%] Built target linker_script_target
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[ 49%] Built target arch__common
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo'
[ 59%] Built target arch__arm__core
[... 720 lines cut ...]
make[3]: Entering directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo.poll'
make[3]: Leaving directory '/home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo.poll'

**====ERROR OUTPUT END====**

**To Reproduce**
Steps to reproduce the behavior:
1. Use Microchip board https://docs.zephyrproject.org/latest/boards/arm/mec15xxevb_assy6853/doc/index.html
2. Use the latest master or the commit 23bdddea0
3. Run Sanitycheck ` sanitycheck --device-testing --device-serial /dev/ttyUSB1 -p mec15xxevb_assy6853 -T /home/maksim/zephyrproject/zephyr/tests/kernel/fifo/fifo_api/ -v -v`
you can also run the sanitycheck using hardware map for the Microchip board -the same behaviour.
4. See the error (Timeout).

**Expected behavior**
Should have output of the running tests and pass that FIFO test succesffully as it was before. Now that FIFO test or any other test not work on the Microchip board due to that error.

**Impact**
Can't use Microchip for the Sanitycheck.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: Fedora 28
 - Toolchain Zephyr SDK 0.10.3
 - 23bdddea0 or the latest master

nashif, 2020-01-02:
The commit 23bddde being the last commit that works is curious. This was in the release phase of 2.1 and is one commit before rc3, nothing in that area changed, so I am not sure how you got to this commit. Are you saying that 2.1 (commit b413223a66ed9e857c67ee21d1e8be4768804dd1) does not work also?

Please try manually and verify that you can build, flash and run those tests without sanitycheck on current master.

nashif, 2020-01-02:
Also, please attach the contents of device.log, handler.log and build.log from  sanity-out/mec15xxevb_assy6853/tests/kernel/fifo/fifo_api/kernel.fifo/

maksimmasalski, 2020-01-03:
@nashif Wrong wiring of the board caused that problem. Before wiring was correct, and then changed, so I missed that wiring can be a root cause too. In sanitycheck PR #21528 I found a bug, so I submitted a PR #21652 to fix it. That bug also caused a Timeout error, but the root cause was another. Please check it.

closed 2020-01-03 by maksimmasalski