zephyrproject-rtos/zephyr #21597: sht3xd build error on olimexino_stm32

boonpin1975, 2019-12-27:
host : Ubuntu 18.04 LTS 
board olimexino_stm32
sensor : sht3xd 

west build  -p auto -b olimexino_stm32 samples/sensor/sht3xd

error messages :

/zephyrproject/zephyr/drivers/sensor/sht3xd/sht3xd.c:228:14: error: 'DT_INST_0_SENSIRION_SHT3XD_BUS_NAME' undeclared here (not in a function); did you mean 'DT_INST_0_ZEPHYR_MMC_SPI_SLOT_BUS_NAME'?
  .bus_name = DT_INST_0_SENSIRION_SHT3XD_BUS_NAME,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              DT_INST_0_ZEPHYR_MMC_SPI_SLOT_BUS_NAME
/zephyrproject/zephyr/drivers/sensor/sht3xd/sht3xd.c:232:18: error: 'DT_INST_0_SENSIRION_SHT3XD_BASE_ADDRESS' undeclared here (not in a function); did you mean 'DT_INST_0_MMIO_SRAM_BASE_ADDRESS'?
  .base_address = DT_INST_0_SENSIRION_SHT3XD_BASE_ADDRESS,
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                  DT_INST_0_MMIO_SRAM_BASE_ADDRESS
In file included from /zephyrproject/zephyr/drivers/sensor/sht3xd/sht3xd.c:7:
/zephyrproject/zephyr/drivers/sensor/sht3xd/sht3xd.c:238:30: error: 'DT_INST_0_SENSIRION_SHT3XD_LABEL' undeclared here (not in a function); did you mean 'DT_INST_0_ST_STM32_SPI_LABEL'?
 DEVICE_AND_API_INIT(sht3xd0, DT_INST_0_SENSIRION_SHT3XD_LABEL,
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/device.h:106:11: note: in definition of macro 'DEVICE_AND_API_INIT'
   .name = drv_name, .init = (init_fn),     \
           ^~~~~~~~
ninja: build stopped: subcommand failed.

ydamigos, 2019-12-27:
Hi @boonpin1975,
Did you use a [DT overlay](https://docs.zephyrproject.org/latest/application/index.html#application-dt)? If not, you should use one. There are some overlay examples for other boards in `samples/sensor/sht3xd`.

carlescufi, 2019-12-28:
Closing as the [answer provided](https://github.com/zephyrproject-rtos/zephyr/issues/21597#issuecomment-569216992) is accurate.

closed 2019-12-28 by carlescufi