zephyrproject-rtos/zephyr #21599: CONFIG_HEAP_MEM_POOL_SIZE and k_malloc, k_free defines doesnot work with nrf51_pca10028

stephenmasih, 2019-12-27:
**Describe the bug**
I my application i am using the board nrf51_pca10028. I the prj.conf file I am defining a CONFIG_HEAP_MEM_POOL_SIZE=1024 for involving a k_malloc() and k_free(). Even when I define a thread in the kernel it doesn't seem to show any output. When I replace the entire thread code in the main thread it works fine in nrf52, not in nrf51. The code works fine with board nrf52_pca10040. But it shows no output in the console with nrf51.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf51_pca10028 --pristine
2. west flash

**Expected behaviour**
There should be dynamic allocation of memory for my linked list in the Bluetooth application.
t_ble_scan() is a thread function and main() thread is also defined.
[image]

carlescufi, 2019-12-28:
@joerchan could you try to reproduce this?

stephenmasih, 2019-12-31:
I tried debugging, any code defined inside a thread. The code gets compiled i am able to get the zephy.hex file. But the code does not work at all. The same code I tried with nrf52, everything seems to be okay.

joerchan, 2020-01-03:
@stephenmasih I have no way to reproduce this issue. Please update with a description on how I can reproduce it. I cannot really do anything with a screenshot.

You are not checking the returned value of k_malloc, you probably should check that it is infact non-zero before using it.

carlescufi, 2020-02-14:
@stephenmasih I am closing this issue since there has been no activity on it for a while. Please reopen with guidance on how to reproduce if this is still a problem.

closed 2020-02-14 by carlescufi