zephyrproject-rtos/zephyr #21601: '!radio_is_ready()' failed

ayworld, 2019-12-27:
assert: '!radio_is_ready()' failed
E: ***** HARD FAULT *****
E: ARCH_EXCEPT with reason 3

**test board:**

 .\nrf\samples\bluetooth\peripheral_hids_mouse\ on nrf51_pca10028 board.

**error info:**

assert: '!radio_is_ready()' failed
E: ***** HARD FAULT *****
E: ARCH_EXCEPT with reason 3

E: r0/a1: 0x00000003 r1/a2: 0x00000023 r2/a3: 0x40001000
E: r3/a4: 0x0001316f r12/ip: 0x0000000a r14/lr: 0x0000253f
E: xpsr: 0x21000011
E: Faulting instruction address (r15/pc): 0x00013174
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling

**version:**

~/ncs/nrf$ git log
commit 0f9ab8f8a62e92749022ae98baf739ca632c0ddf (HEAD -> master, origin/master)
Author: Ulf Magnusson <Ulf.Magnusson@nordicsemi.no>
Date: Sat Dec 21 05:17:51 2019 +0100

doc: Fix pylint warnings that fail CI

pylint warnings now cause failures in CI. Fix some bad indentation, a
bare 'except:' ('except Exception:' avoids catching Ctrl-C and the
like), and suppress an unused-variable warning about a 'tags' variable
that gets created by Sphinx.

**build:**

 west build -d build_ble_hid_test -b nrf51_pca10028  .\nrf\samples\bluetooth\peripheral_hids_mouse\

**Serial output:**

*** Booting Zephyr OS build v2.0.99-ncs1-rc1-1160-gf2d732b7fa5d ***
Starting Bluetooth Peripheral HIDS mouse example
I: 8 Sectors of 1024 bytes
I: alloc wra: 0, 3a0
I: data wra: 0, 67
I: HW Platform: Nordic Semiconductor (0x0002)
I: HW Variant: nRF51x (0x0001)
I: Firmware: Standard Bluetooth controller (0x00) Version 2.1 Build 0
I: No ID address. App must call settings_load()
Bluetooth initialized
I: Saving ID
I: Identity: d6:f6:55:1d:ee:0f (random)
I: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1
I: LMP: version 5.1 (0x0a) subver 0xffff
Regular advertising started
Connected 43:6a:4a:a5:3f:dc (random)
Regular advertising started
Disconnected from 43:6a:4a:a5:3f:dc (random) (reason 19)
Regular advertising started
Connected 43:6a:4a:a5:3f:dc (random)
Regular advertising started
Passkey for 43:6a:4a:a5:3f:dc (random): 680091
Press Button 1 to confirm, Button 2 to reject.
Numeric Match, conn 0x2000051c
Security changed: 43:6a:4a:a5:3f:dc (random) level 4
Pairing completed: 6c:b7:49:7c:5e:07 (public), bonded: 1
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): down
button_changed(): down
button_changed(): down
button_changed(): down
button_changed(): down
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): up
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): up
button_changed(): right
button_changed(): right
button_changed(): up
assert: '!radio_is_ready()' failed
E: ***** HARD FAULT *****
E: ARCH_EXCEPT with reason 3

E: r0/a1: 0x00000003 r1/a2: 0x00000023 r2/a3: 0x40001000
E: r3/a4: 0x0001316f r12/ip: 0x0000000a r14/lr: 0x0000253f
E: xpsr: 0x21000011
E: Faulting instruction address (r15/pc): 0x00013174
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20002078 (unknown)
E: Resetting system
*** Booting Zephyr OS build v2.0.99-ncs1-rc1-1160-gf2d732b7fa5d ***
Starting Bluetooth Peripheral HIDS mouse example
I: 8 Sectors of 1024 bytes
I: alloc wra: 0, 378
I: data wra: 0, ec
I: HW Platform: Nordic Semiconductor (0x0002)
I: HW Variant: nRF51x (0x0001)
I: Firmware: Standard Bluetooth controller (0x00) Version 2.1 Build 0
I: No ID address. App must call settings_load()
Bluetooth initialized
I: Saving ID
I: Identity: d6:f6:55:1d:ee:0f (random)
I: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1
I: LMP: version 5.1 (0x0a) subver 0xffff
Direct advertising to 6c:b7:49:7c:5e:07 (public) started
Direct advertising to 6c:b7:49:7c:5e:07 (public) timed out
Regular advertising started
button_changed(): up
button_changed(): right
button_changed(): right
Connected 6c:b7:49:7c:5e:07 (public)
Advertising failed to start (err -5)
Security changed: 6c:b7:49:7c:5e:07 (public) level 4

cvinayak, 2020-01-06:
Indicates Radio ISR latency, will analyse and post my comments here. Meanwhile, could you please bisect to an older release, say v1.14 may be, to find when it is ok and broken after a commit.

jhedberg, 2020-02-18:
@cvinayak have you had a chance to take another look at this?

cvinayak, 2020-02-20:
@jhedberg No, I have not profiled the cause of the latency. This seem will have to be postponed after 2.2 due to lack of time.

PR #23151 (2020-02-28): Bluetooth: controller: Add Kconfig for Optimize for Speed
Add Kconfig option to support building the controller
optimized for speed.

Fixes #21601.
commit f972fb05 (2020-02-28): Bluetooth: controller: Add Kconfig for Optimize for Speed

Add Kconfig option to support building the controller
optimized for speed.

Fixes #21601.
closed 2020-02-28 by jhedberg