zephyrproject-rtos/zephyr #21607: BME680 Sensor is not building

md-jamal, 2019-12-27:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
$ west build -b sam_e70_xplained samples/sensor/bme680

**Expected behavior**
Build failed. From one of the issues, it says to update device tree . Can you please provide an sample device tree for bme680

Adding this, didn't work

bme680@77 {
          compatible = "bosch,bme680";
          reg = <0x77>;
};

**Screenshots or console output**
/home/abc/zephyrproject/zephyr/samples/sensor/bme680/src/main.c:14:42: error: 'DT_INST_0_BOSCH_BME680_LABEL' undeclared (first use in this function); did you mean 'DT_INST_0_SOC_NV_FLASH_LABEL'?
  struct device *dev = device_get_binding(DT_INST_0_BOSCH_BME680_LABEL);
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                          DT_INST_0_SOC_NV_FLASH_LABEL

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 
 - Commit SHA or Version used: 2.1.99

**Additional context**
Add any other context about the problem here.

ulfalizer, 2019-12-27:
CC @BoschSensortec 

Not familiar with this device, but try making it

```
bme680@77 {
	compatible = "bosch,bme680";
	reg = <0x77>;
	label = "BME680";
};
```

Doing a `git grep` for some of the other compatible strings you'll find in the bindings in `dts/bindings/sensor/` might be helpful too, e.g. `bosch,bmg160`.

md-jamal, 2019-12-27:
Added the following to sam_e70_xplained.overlay
```
bme680@77 {
	compatible = "bosch,bme680";
	reg = <0x77>;
	label = "BME680";
};
```
I get the following error:

```
Error: sam_e70_xplained.dts.pre.tmp:403.1-10 syntax error
FATAL ERROR: Unable to parse input tree
CMake Error at /home/linuxtrainer/zephyrproject/zephyr/cmake/dts.cmake:183 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/linuxtrainer/zephyrproject/zephyr/cmake/app/boilerplate.cmake:460 (include)
  CMakeLists.txt:8 (include)

```

pabigot, 2019-12-27:
The node has to be a child of an I2C controller.  E.g.:
```
&i2c0 {
  bme60@77 {
    label = "bme680";
    compatible = "bosch,bme680";
    reg = <0x77>;
  };
};
```

md-jamal, 2019-12-27:
```
sam_e70_xplained.dts.pre.tmp:342.3-16: Warning (reg_format): /bme680@77:reg: property has invalid length (4 bytes) (#address-cells == 1, #size-cells == 1)
sam_e70_xplained.dts_compiled: Warning (pci_device_reg): Failed prerequisite 'reg_format'
sam_e70_xplained.dts_compiled: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
sam_e70_xplained.dts_compiled: Warning (simple_bus_reg): Failed prerequisite 'reg_format'
sam_e70_xplained.dts_compiled: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
sam_e70_xplained.dts_compiled: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
devicetree error: 'reg' property in <Node /bme680@77 in 'sam_e70_xplained.dts.pre.tmp'> has length 4, which is not evenly divisible by 8
CMake Error at /home/linuxtrainer/zephyrproject/zephyr/cmake/dts.cmake:203 (message):
  new extractor failed with return code: 1
Call Stack (most recent call first):
  /home/linuxtrainer/zephyrproject/zephyr/cmake/app/boilerplate.cmake:460 (include)
  CMakeLists.txt:8 (include)

```

md-jamal, 2019-12-28:
It worked...

carlescufi, 2019-12-28:
I believe the issue is that you need to add a bme680 instance in your device tree file.

carlescufi, 2019-12-28:
Try something like this:

```
&i2c1 {
	compatible = ...

	bme680@76 {
		compatible = "bosch,bme680";
		label = "BME680";
		reg = <0x76>;
	};
};
```

nashif, 2020-01-20:
seems to be answered, closing.

closed 2020-01-20 by nashif