zephyrproject-rtos/zephyr #21611: IS25LP032D-JNLE Flash support

md-jamal, 2019-12-28:
Hi All,

I have a board on which I have http://www.issi.com/WW/pdf/25LP-WP032D.pdf the following SPI Flash connected.

What all changes are required for supporting this device on Zephyr.

I see there is already an sample application present in samples/drivers/spi_flash/. I want to make it compatible for all the application.

One more question, is there any sample application which just tests SPI Communication

pabigot, 2019-12-29:
Several Zephyr in-tree boards have a related device.  It should be supported with the spi-nor flash driver by adding something like this under the corresponding flash device in devicetree:
```
&spi0 {
        is25wp032@0 {
                compatible = "issi,is25wp032", "jedec,spi-nor";
                size = <33554432>;
                label = "IS25WP032";
                reg = <0>;
                spi-max-frequency = <66000000>;
                status = "okay";
                jedec-id = [9d 70 16];
        };
};
```
You should be able to test this with `samples/drivers/spi_flash` by adding a board config for your target in the `boards` subdirectory that sets `CONFIG_SPI_NOR=y`.

I'm unaware of any tests for spi communication that don't involve having a spi device, except possibly `tests/drivers/spi/spi_loopback` which might require MISO/MOSI cross-connect.

md-jamal, 2019-12-30:
I performed the following steps.

Step1 : Created sam_e70_xplained.conf file in zephyrproject/zephyr/samples/drivers/spi_flash/boards with the following content.

#
# Copyright (c) 2019 Peter Bigot Consulting, LLC
#
# SPDX-License-Identifier: Apache-2.0
#

CONFIG_SPI_NOR=y

Step2 : Created overlay file sam_e70_xplained.overlay in zephyrproject/zephyr/samples/drivers/spi_flash with the following content

&spi0 {
        is25wp032@0 {
                compatible = "issi,is25wp032", "jedec,spi-nor";
                size = <33554432>;
                label = "IS25WP032";
                reg = <0>;
                spi-max-frequency = <66000000>;
                status = "okay";
                jedec-id = [9d 70 16];
        };
};

Step 3: $ west build -b sam_e70_xplained samples/drivers/spi_flash/
Step 4: $ west flash

I get the following:
*** Booting Zephyr OS build zephyr-v2.1.0-581-g4e135d76a3c2  ***
                                        
JEDEC SPI-NOR SPI flash testing         
==========================              
SPI flash driver IS25WP032 was not found!

pabigot, 2019-12-30:
Presumably the flash is on a shield or some other board, since the SAM E70 doesn't have this device.

Try adding the following to `prj.conf` to see if there's a clue why it doesn't work.  If the answer isn't obvious, you may need to modify `drivers/flash/spi-nor.c` to add diagnostic output.

```
CONFIG_LOG=y
CONFIG_FLASH_LOG_LEVEL_DBG=y
```

md-jamal, 2019-12-30:
I did the above steps and here is the log

```
*** Booting Zephyr OS build zephyr-v2.1.0-581-g4e135d76a3c2  ***

JEDEC SPI-NOR SPI flash testing
==========================
SPI flash driver IS25WP032 was not found!
[00:00:00.000,000] <wrn> soc: CIDR mismatch: chip = 0xa1020e01 vs HAL = 0xa1020e00
[00:00:00.000,000] <inf> spi_sam: CS control inhibited (no GPIO device)

```

md-jamal, 2019-12-30:
I have added log in spi_nor_read_id,,

```
        if (memcmp(flash_id->id, buf, SPI_NOR_MAX_ID_LEN) != 0) {
                LOG_INF("%s enodev, buf[0]:%d\t buf[1]:%d\tbuf[2]:%d\n", __func__, buf[0],
                buf[1], buf[2]);
                return -ENODEV;
        }
```
buf[0] = 0, buf[1] = 0, buf[2] = 0 .
What does this mean

pabigot, 2019-12-30:
> SPI flash driver IS25WP032 was not found!
> [00:00:00.000,000] <wrn> soc: CIDR mismatch: chip = 0xa1020e01 vs HAL = 0xa1020e00

Somebody who knows Atmel parts would have to answer this, but it looks like a hardware configuration problem

> [00:00:00.000,000] <inf> spi_sam: CS control inhibited (no GPIO device)
> ```

It also looks like there's no chip-select for the SPI bus you've selected.

If the device isn't being accessed, the ID printed as 0, 0, 0 is not relevant.

md-jamal, 2019-12-30:
What do you mean by no chip select for SPI bus?

pabigot, 2019-12-30:
SPI devices require a chip select signal to indicate which device the controller wants to address.  The diagnostic indicates the devicetree or pinmux configuration for this bus does not identify a chip select.  This is not something I can help with as I'm not very familiar with the Atmel hardware.

md-jamal, 2019-12-30:
[image]

Chip select line is PD26

md-jamal, 2019-12-30:
Do we need to provide cs-gpios in device tree?

md-jamal, 2020-01-13:
How can we provide chip select pin in device tree?

pabigot, 2020-01-16:
The Zephyr board support for SAM E70 doesn't seem to have SPI infrastructure in it.  You will probably have to add it.  I would use `atsamd20_xpro` as an example.  You will need to add SPI-related options to `sam_e70_xplained_defconfig`, and to add configuration in the pinmux.c file as well.  If SAM E70 uses pinmux you won't need a `cs-gpios` property.

I don't see any other boards that use `CONFIG_SPI_SAM` (as opposed to `CONFIG_SPI_SAM0`) so there may be a lot of gaps to fill.

md-jamal, 2020-01-17:
But the board support page says 'spi is supported'
[image]

Am i missing anything?

pabigot, 2020-01-17:
The diagnostics are:

    [00:00:00.000,000] <wrn> soc: CIDR mismatch: chip = 0xa1020e01 vs HAL = 0xa1020e00

I still don't know what that is.

    [00:00:00.000,000] <inf> spi_sam: CS control inhibited (no GPIO device)

Looking into this I see that it's supposed to use a hard-coded chip-select specified as a pinmux outside of devicetree.  That external specification doesn't provide a GPIO device, so the SPI infrastructure refuses to use it.  The pinmux function appears not to be initialized.  This all went in under #10620.  I'm not convinced it ever worked.

I've gotta muck with this area anyway, so let me see what can be done.

tbursztyka, 2020-01-17:
@pabigot I agree the info message is misleading, it should be understood "CS GPIO control unused". I think no message at all would be just better in this case. Now about spi_sam.c's internal configuration of its CS pin, there is no better way to check but with a scope.

md-jamal, 2020-01-17:
I am still unable to understand what needs to be implemented.
tests/drivers/spi/spi_loopback successfully works when i loopback MOSI to MISO. Does this not mean SPI is working fine?

md-jamal, 2020-01-18:
I have added to `samples/drivers/spi_flash/sam_e70_xplained.overlay`
`cs-gpios = <&portd 25 0>;`

In drivers/flash/spi_nor.c, it returns at this stage in spi_nor_configure

```
if (!data->cs_ctrl.gpio_dev) {
                return -ENODEV;
        }

```

md-jamal, 2020-01-18:
After adding `CONFIG_GPIO=y` it worked. Thanks everyone. I also need to update the jdec id

[image]

closed 2020-06-25 by MaureenHelm