zephyrproject-rtos/zephyr #21617: Allow per module prj.conf

hadrienk, 2019-12-28:
**Is your enhancement proposal related to a problem? Please describe.**

I am using mcuboot and spm. I need to manually update the .config under `build/spm/zephyr/.config` when the build directory is cleaned.

**Describe the solution you'd like**

Use a special configuration file name convention: 
`prj_mcuboot.conf`
`prj_spm.conf`

This should be compatible with the board configuration `prj_spm_YOURBOARD.conf`

carlescufi, 2019-12-29:
@hadrienk  thanks for the feedback. spm is not part of Zephyr, but rather only present in Nordic's Zephyr-based SDK. Could you edit your proposal so that it doesn't mention spm and instead focuses on mcuboot?

hadrienk, 2020-01-03:
@carlescufi sure thing.

carlescufi, 2020-01-09:
@ulfalizer could you comment on this?

SebastianBoe, 2020-01-16:
In NCS one can specify the mcuboot prj.conf like this:

https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/applications/nrf_desktop/CMakeLists.txt#L34

In Zephyr upstream it doesn't make sense to put a child image's build dir inside of the parent image's build dir so this problem does not exist.

Please re-open if more clarification is needed.

closed 2020-01-16 by SebastianBoe

hadrienk, 2020-01-31:
Thanks. Got it to work. If anyone want to do the same, remember to set the image_CONF_FILE **before** the `include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake)`
```
set(mcuboot_CONF_FILE "${CMAKE_CURRENT_SOURCE_DIR}/mcuboot.conf")
include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake)
```