zephyrproject-rtos/zephyr #21618: CI failing to complete tests

pabigot, 2019-12-29:
As reported on Slack 2019-12-23T15:55Z the documentation, gitlint, identity/emails, and Kconfig tests are failing to complete on multiple PRs.  Resubmitting does not seem to have any effect.  See https://github.com/zephyrproject-rtos/zephyr/pulls and note all the orange dots where there should be a :heavy_multiplication_x: or a :heavy_check_mark:.

jhedberg, 2020-01-02:
This still doesn't seem to be resolved and in practice means that nothing can be merged at the moment.

stephanosio, 2020-01-02:
Need to revert the following commit and keep using the docker image v0.9.1 until we figure out why the CI is not happy with the new CI image (probably Python 3.6-related issue ...):

https://github.com/zephyrproject-rtos/ci-pipelines/commit/6bf203c057475d5993b12f523e6b10b41bce0231

nashif, 2020-01-02:
should be resolved now.

closed 2020-01-02 by nashif