zephyrproject-rtos/zephyr #21633: Support for MQTT 5

nullr0ute, 2020-01-01:
**Is your feature request related to a problem? Please describe.**
MQTT v5 is the latest version of the MQTT protocol.

**Describe the solution you'd like**
Adding v5 support to the existing MQTT protocol support so that there is one pub/sub client to support all version of the protocol.

**Additional context**
Version 5 of the MQTT protocol is published by OASIS at the following link:
https://docs.oasis-open.org/mqtt/mqtt/v5.0/mqtt-v5.0.html

The Eclipse PAHO project has open source client reference implementations supporting version 5 in a number of languages:
http://www.eclipse.org/paho/

carlescufi, 2020-03-31:
@rlubos any plans to implement this in the near/mid term?

rlubos, 2020-04-01:
@carlescufi Not on my schedule, at least for now. We might consider it though as it seems like a valuable improvement.

Linutux, 2020-11-02:
I think MQTT 5 is particularly important for IoT devices, as it has some  unique features, like "Maximum message size" or "Negative Acknowledgements", which tell the client why a massage could not be delivered.

Aeted-RC, 2021-08-31:
Looking forward to implementing this plan, it is very meaningful

rlubos, 2022-02-17:
As for Nordic, it's not on our schedule (yet), but other contributions are welcome.

plskeggs, 2023-09-19:
@rlubos Amazon AWS now supports MQTT 5: https://docs.aws.amazon.com/iot/latest/developerguide/mqtt.html#mqtt5. The maximum packet size and receive maximum properties of the CONNECT packet are desirable.

tka85, 2024-01-30:
Any upddate on this? Mqtt 5.0 is light years ahead of 3.1.1. They have added significant features that address all of the most common IoT pain points.

rlubos, 2024-01-30:
> Any upddate on this?

No progress according to my knowledge.

iot-guy-col, 2024-06-18:
Any update on supporting MQTT 5? Is anyone using 3rd party libraries on Nordic microcontrollers?

rlubos, 2024-06-19:
> Any update on supporting MQTT 5?

No update.

rlubos, 2025-03-19:
For anyone interested in the topic: https://github.com/zephyrproject-rtos/zephyr/pull/87354

closed 2025-03-26 by kartben