zephyrproject-rtos/zephyr #21635: sht3xd error -5 on olimexino_stm32

boonpin1975, 2020-01-02:
hi , 

 I could build the sht3xd on olimexino_stm32 after adding the overlay file , thanks to @ydamigos ,
I am connected to PB6 and PB7 on the olimexino_stm32 to sht3xd . the reading for console after flashing the binary : 
�*** Booting Zephyr OS build zephyr-v2.1.0-581-g4e135d76a3c2  ***               
SHT3XD: 24.32 Cel ; 54.02 %RH                                                   
SHT3XD: failed: -5 
it did not run the loop , as mention in the https://docs.zephyrproject.org/latest/samples/sensor/sht3xd/README.html

thanks !

pabigot, 2020-01-02:
Try adding this to `prj.conf` and rebuilding, to see if there's any explanation why the read fails:
```
CONFIG_LOG=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
CONFIG_I2C_LOG_LEVEL_DBG=y
```

boonpin1975, 2020-01-06:
I added this line and rebuild , but now there is no more error -5 
SHT3XD: 25.79 Cel ; 50.17 %RH
�*** Booting Zephyr OS build zephyr-v2.1.0-581-g4e135d76a3c2  ***
SHT3XD: 25.73 Cel ; 50.28 %RH
SHT3XD: 25.72 Cel ; 50.69 %RH
SHT3XD: 25.70 Cel ; 50.76 %RH
SHT3XD: 25.69 Cel ; 50.79 %RH
SHT3XD: 25.65 Cel ; 50.89 %RH
SHT3XD: 25.59 Cel ; 50.88 %RH
SHT3XD: 25.55 Cel ; 50.85 %RH

sslupsky, 2020-01-06:
That looks like you had (or have) a CRC error.  I had issues with the i2c driver for sam0 and the stop bit and saw this error a few times.  You appear to be receiving some valid data so maybe you are not receiving the last byte and the CRC check fails?  That might also explain why it was not looping.

pabigot, 2020-06-29:
Closing due to inactivity and inability to reproduce.

closed 2020-06-29 by pabigot