zephyrproject-rtos/zephyr #21637: sanitycheck failed issue in parallel running.

chen-png, 2020-01-02:
**Describe the bug**
use the sanitycheck to run all tests with four boards, it failed with error **"generated an exception: [Errno 24] Too many open files**", i found this error is caused by the code "log_out_fp = open(self.log, "wt")" in DeviceHandler, the system limits the max open files to 1024.
then I modified the /etc/secruity/limits.conf to set the limited open files to 2048, and run sanitycheck again, it failed with "**ValueError: filedescriptor out of range in select()**", caused by code" readable, _, _ = select.select(readlist, [], [], self.timeout)

**To Reproduce**
Steps to reproduce the behavior:
1. connect four boars to pc and run all tests use sanitycheck
2. sanitycheck  --generate-hardware-map map4.yml
4. sanitycheck  --hardware-map  map4.yml  --device-testing  -T  tests -v
4. See error

**Screenshots or console output**

**Exception in thread Thread-635:**
Traceback (most recent call last):
  File "/usr/lib64/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/chenpeng/zephyrproject/zephyr/scripts/sanitycheck", line 628, in monitor_serial
    readable, _, _ = select.select(readlist, [], [], self.timeout)
**ValueError: filedescriptor out of range in select()**

INFO    - 666/720 mimxrt1050_evk            tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations FAILED N/A (device 8.871s)
ERROR   - see: sanity-out/mimxrt1050_evk/tests/arch/arm/arm_interrupt/arch.interrupt.no_optimizations/build.log
Exception in thread Thread-636:
Traceback (most recent call last):
  File "/usr/lib64/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/chenpeng/zephyrproject/zephyr/scripts/sanitycheck", line 628, in monitor_serial
    readable, _, _ = select.select(readlist, [], [], self.timeout)
ValueError: filedescriptor out of range in select()

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: 4e135d76a3c2

aescolar, 2020-01-02:
CC @nashif

nashif, 2020-01-02:
duplicate of #20974

PR #21684 (2020-01-06): Sanitycheck fixes, enhancements
- close pipe file handles
- More debug messages
- add Digilent to device discovery

Fixes #20974
Fixes #21637
commit bdffb889 (2020-01-06): sanitycheck: close pipe file handles

Make sure we close the pipe file handles after we are done, otherwise we
will end up with too many open file descriptors and crash...

Fixes #20974
Fixes #21637
closed 2020-01-06 by nashif