zephyrproject-rtos/zephyr #21647: Support using the shell to mount littlefs filesystems in partitions other than 'storage'

SebastianBoe, 2020-01-02:
The shell can be used to mount filesystems and read their contents.

But even though littlefs could at runtime be initialized at any partition, the shell-integration in subsys/fs/shell.c is assuming that the littlefs is always mounted in the 'storage' partition.

littlefs is not always mounted at the storage partition, for instance when the settings subsystem is using the storage partition, the littlefs filesystem will have to be placed in a different partition.

subsys/fs/shell.c should be updated to allow mounting littlefs on any partition.

SebastianBoe, 2020-01-02:
The code that is hardcoding the assumption that littlefs is in the storage partition is located in subsys/fs/shell.c and looks like this:

```
static struct fs_mount_t littlefs_mnt = {
	.type = FS_LITTLEFS,
	.fs_data = &lfs_data,
	.storage_dev = (void *)DT_FLASH_AREA_STORAGE_ID,
};

```

sslupsky, 2022-07-18:
There is another problem as well, the shell integration also forces the default littlefs configuration:

https://github.com/zephyrproject-rtos/zephyr/blob/03912c23e380c9e36a7ce40cc061f58720f67e73/subsys/fs/shell.c#L35

So if you are using a custom configuration, it is not compatible.