zephyrproject-rtos/zephyr #21648: improve documentation on meta-IRQ threads

pabigot, 2020-01-02:
Related to #21185 there is a distinct lack of documentation on meta-IRQ threads and how they can be used safely.  They are not mentioned at all in the discusson of [thread priorities](https://docs.zephyrproject.org/latest/reference/kernel/threads/index.html#thread-priorities).

For example, it appears that somebody who's using `K_PRIO_COOP(0)` to create a cooperative thread might be surprised to learn that when `CONFIG_NUM_METAIRQ_PRIORITIES` is positive this thread can now pre-empt other "cooperative" threads.

nashif, 2020-01-02:
You mean in addition to this https://docs.zephyrproject.org/latest/reference/kernel/scheduling/index.html?highlight=meta#meta-irq-prioritie and this https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NUM_METAIRQ_PRIORITIES.html#cmdoption-arg-config-num-metairq-priorities?

pabigot, 2020-01-02:
Yes.

andrewboie, 2020-01-02:
@pabigot this being an open source project the fastest path to success is if you would send a PR with the changes you want.

nashif, 2021-05-28:
@laurenmurphyx64  Can you please look into this?

PR #37885 (2021-08-24): doc: kernel: move meta-irq doc to threads
Move this to where priorities are being discussed to keep things in
context and to have all priority types documented in 1 place.

Fixes #21648
commit eef51a89 (2021-08-24): doc: kernel: move meta-irq doc to threads

Move this to where priorities are being discussed to keep things in
context and to have all priority types documented in 1 place.

Fixes #21648
closed 2021-08-24 by nashif