zephyrproject-rtos/zephyr #21650: _TEXT_SECTION_NAME_2 on ARM 32-bit

andrewboie, 2020-01-02:
**Describe the bug**
ARM platforms define a _TEXT_SECTION_NAME_2 which ends up in the final binary.
This isn't right. We just want one text section, if stuff needs to be aligned internally we can do that within the text section itself.

Delete the exception for _TEXT_SECTION_NAME_2 in scripts/sanitycheck.

ioannisg, 2020-01-10:
As far as I could see this has little to do with alignment @andrewboie , it is rather related to the code relocation work done in #10892

I wonder if we could simply rename the initial TEXT Section that contains the vector table to something else.

ioannisg, 2020-01-10:
>I wonder if we could simply rename the initial TEXT Section that contains the vector table to something else.

@wentongwu do you think we can do the above?

wentongwu, 2020-01-13:
> > I wonder if we could simply rename the initial TEXT Section that contains the vector table to something else.
> 
> @wentongwu do you think we can do the above?

@ioannisg I think we can rename this section, maybe vector_table_section.

ioannisg, 2020-01-13:
> > > I wonder if we could simply rename the initial TEXT Section that contains the vector table to something else.
> > 
> > 
> > @wentongwu do you think we can do the above?
> 
> @ioannisg I think we can rename this section, maybe _vector_table_section.

@wentongwu @andrewboie  I am ok with that. Actually, at least 2 other architectures, Risc-V and Nios2 have a similar thing, that is, several sections at the beginning of ROM (vector, reset exception section) followed by the text section. 
This means that __image_rom_start and __image_text_start will be different. The whole thing seems trivial; I'll send a PR and we can take the discussion there. I would like to make sure we are not breaking anything by doing that.

andrewboie, 2020-01-13:
@ioannisg sounds right -- what we really have here is a special rodata section immediately before the actual program text.

Thinking out loud: could we just reverse the order of the text and rodata sections? So what we have now is:

```
vector table and other special rom
text section
.ARM.extab (more special ROM)
.ARM.exidx (more special ROM)
rodata section
```

Could we just do

```
rodata section (containing vector table + extab/exidx + zephyr rodata)
text section
```

ioannisg, 2020-01-16:
#21874 fixes this for Cortex-M, I would prefer if @stephanosio provides the fix for Cortex-R.

stephanosio, 2020-02-08:
@ioannisg I will look into this on Monday.

closed 2020-02-12 by ioannisg