zephyrproject-rtos/zephyr #21660: Sample projects do not build for Nucleo WB55RG

r2r0, 2020-01-03:
Samples i.e. hello_world, basic/blinky, basic/threads cannot be build for Nucleo WB55RG board.
It looks that stm32wbxx.h file is missing.
The same problem when using west build or "manual" cmake.

**Steps to reproduce the behavior**
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nucleo_wb55rg ../zephyr/samples/basic/threads
3. ninja
4. See error

**Expected behavior**
Clean build of samples

**Impact**
Showstopper.

**Screenshots or console output**
hid@devel:~/lpm/zephyr/build$ cmake -GNinja -DBOARD=nucleo_wb55rg ../samples/basic/threads
Zephyr version: 2.1.99
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6") 
-- Selected BOARD nucleo_wb55rg
-- Found west: /home/hid/.local/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
-- Loading /home/hid/lpm/zephyr/boards/arm/nucleo_wb55rg/nucleo_wb55rg.dts as base
Devicetree configuration written to /home/hid/lpm/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/hid/lpm/zephyr/Kconfig
Loaded configuration '/home/hid/lpm/zephyr/boards/arm/nucleo_wb55rg/nucleo_wb55rg_defconfig'
Merged configuration '/home/hid/lpm/zephyr/samples/basic/threads/prj.conf'
Configuration saved to '/home/hid/lpm/zephyr/build/zephyr/.config'
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/hid/.cache/zephyr
CMake Warning at ../../../CMakeLists.txt:1502 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /home/hid/lpm/zephyr/build
hid@devel:~/lpm/zephyr/build$ ninja
[3/120] Preparing syscall dependency handling

[7/120] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.1.0-632-gb35d414916c3 -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../kernel/include -I../arch/arm/include -I../include -I../include/drivers -Izephyr/include/generated -I../soc/arm/st_stm32/stm32wb -I../drivers -I../ext/hal/cmsis/Core/Include -isystem ../lib/libc/minimal/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -imacros/home/hid/lpm/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/home/hid/lpm/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/hid/lpm/zephyr/samples/basic/threads=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/hid/lpm/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj   -c /home/hid/lpm/zephyr/arch/arm/core/offsets/offsets.c
In file included from ../include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from ../include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h:10,
                 from ../include/arch/arm/aarch32/cortex_m/mpu/arm_mpu.h:13,
                 from ../include/arch/arm/aarch32/arch.h:248,
                 from ../include/arch/cpu.h:17,
                 from ../include/kernel_includes.h:34,
                 from ../include/kernel.h:17,
                 from /home/hid/lpm/zephyr/arch/arm/core/offsets/offsets.c:25:
../soc/arm/st_stm32/stm32wb/soc.h:24:10: fatal error: stm32wbxx.h: No such file or directory
 #include <stm32wbxx.h>
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

**Environment**
 - OS: Linux (Ubuntu 18.04)
 - Toolchain: Zephyr SDK
 - Commit SHA: b35d414916c32e66ed1422c02be74a4e7a5feb7f (master at time of this report creation)

jfischer-phytec-iot, 2020-01-03:
Can not reproduce for hello_world and basic/blinky. `west update` ?

r2r0, 2020-01-03:
After manual "west update" it works.
It looks that for some reason script used for testing master branch with WB55 board silently failed with this step. I manually checked only build steps.
Sorry for noise.

closed 2020-01-03 by r2r0