zephyrproject-rtos/zephyr #21669: Bluetooth: Consider flash wear when deleting bond information.

joerchan, 2020-01-03:
**Is your enhancement proposal related to a problem? Please describe.**
Deleting multiple bonds trigger many calls to settings_delete in BT settings. This could potentially cause unnecessary flash performance and wear.

**Describe the solution you'd like**
Submit multiple deletion as a single update to the settings backend. This could be done similar to how BT identities are stored.

**Describe alternatives you've considered**
Leave as is. Might be a non-issue if this is rarely done/used.

**Additional context**
N/A

joerchan, 2020-01-03:
CC: @jsiverskog

joerchan, 2024-02-09:
Who's impersonating the Zephyr Bot?

I opened up this issue on behalf of someone else, either on discord / slack / mailing list / github issue. I don't remember.

I'm happy to allow this issue to be closed as a won't do feature. I haven't seen this arise as an issue, and was only given as a potential problem from what I remember.

jsiverskog, 2024-02-09:
> EDIT: From the CC I'm guessing @jsiverskog was the one that pointed this out.

i'm fine with closing this issue.

closed 2024-02-09 by jori-nordic