zephyrproject-rtos/zephyr #21670: Keep device structures in ROM

nordic-krch, 2020-01-03:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, `struct device` and `struct device_config` are created in RAM for each device. This means 24 bytes of RAM for each device. Hello world sample on nrf52 is using 192 bytes for that. It seems like most or even all fields are constant in those structures. I've found following runtime uses: 
- `driver_api` field is set to NULL when initialization fails. Init flag could be moved to dedicated struct stored in RAM
- some drivers are writing to `driver_api` pointer in init function but it seems like it can be easily fixed 

**Describe the solution you'd like**
Make device structures const. Keep ram control block for runtime data (e.g. init result). 20 bytes of RAM per device can be saved. It means easily >500 bytes saving for even small application. Especially that `SYS_INIT` macro is using device structure underneath.

nordic-krch, 2020-01-03:
@nashif has that been considered before?

sslupsky, 2020-01-07:
I like this idea.  Can some of the device pm settings be stored in ROM?  They take a LOT of RAM when you turn on device pm.

nashif, 2020-01-16:
> @nashif has that been considered before?

dont remember, but this was a design decision back when this was introduced, the relevant parts from the original RFC (from 2015):

```
Each object type will be in its own section in OS image. I am going to
in reverse order IP block specific -> Generic object to try to avoid
forward references.

First is the IP block specific config block. This config structure
contains the configuration parameters specific to a given driver. This
structure is defined by the driver writer.  This structure is
initialized by platform specific init code. This can be hardcoded or
come from configuration variables depending on the needs of the
platform.  This object is the leaf node of the device model tree.
The initialized structures are place in a ROM section of the
image.

Next is the generic configuration object.  This object contains the
name that the driver is exposing to the system, a reference to the
IP block specifc config block and the address of the driver init()
function.  This object type will also be placed in a ROM section
of the image.

Last there is the top level device structure.  This is the structure
that will be passed as an argument to *all* driver entry points. These
structures will be located in *RAM*.  The device structure has a pointer
to the device configuration information, a pointer for the driver to
store a pointer to its context data and a pointer to API functions
for the driver.  Each type of driver (UART, SPI, I2C) will support a
type specific API. The driver will fill in the pointer to the structure
containing the function pointers to its API functions during driver
init().  These structures will placed into the RAM section in init
level order via the __define_initcall() macro.
```

jfischer-phytec-iot, 2020-01-16:
> Currently, `struct device` and `struct device_config` are created in RAM for each device.

`struct device_config` will be placed in the ROM section [".devconfig.init"](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/linker/common-rom.ld#L73)

nordic-krch, 2020-01-21:
`ram_report` shows that it's in ram
```
drivers                                                                                  192     3.77%
  clock_control                                                                           48     0.94%
    nrf_power_clock.c                                                                     48     0.94%
      __config_clock_nrf                                                                  12     0.24%
      __device_clock_nrf                                                                  12     0.24%
      data                                                                                24     0.47%
  console                                                                                 28     0.55%
    uart_console.c                                                                        28     0.55%
      __config_sys_init_uart_console_init0                                                12     0.24%
      __device_sys_init_uart_console_init0                                                12     0.24%
```

nashif, 2020-08-04:
should be addressed already in other RFCs and PRs.

closed 2020-08-04 by nashif