zephyrproject-rtos/zephyr #21677: [Coverity CID :206388] Unrecoverable parse warning in subsys/cpp/cpp_new.cpp

zephyrbot, 2020-01-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/53392ad403db3a3c63e5802e620ead569306b978/subsys/cpp/cpp_new.cpp#L9

Category: Parse warnings
Function: ``
Component: Other
CID: [206388](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=206388)

Details:

```
3      *
4      * SPDX-License-Identifier: Apache-2.0
5      */
6    
7     #include <stdlib.h>
8    
>>>     CID 206388:  Parse warnings  (PARSE_ERROR)
>>>     first parameter of allocation function must be of type "size_t"
9     void* operator new(size_t size)
10     {
11         return malloc(size);
12     }
13    
14     void* operator new[](size_t size)
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

pabigot, 2020-01-03:
I don't see what's to be done about this other than marking it to be ignored.  The first parameter of the allocation is obviously of type `size_t`.

Clearly the first parameter of the allocation is, in fact, of type size_t.  It may be that Coverity is finding a version of `<stdlib.h>` that doesn't put that typedef in the right namespace, but that's not something we can fix in mainline Zephyr: even for minimal libc it comes from `<stddef.h>` which comes from the toolchain.

@nashif This needs to be assigned to somebody who can mark this as ignore in Coverity. I don't have that authority.

pabigot, 2020-03-11:
Marked ignored in Coverity with this comment:

The required argument type is std::size_t rather than ::size_t, which may be what Coverity is diagnosing.  This particular module is built only when we don't have libcxx, so can't switch to <cstdlib> or <cstddef> which would provide the proper name.

Assume that no toolchain will use a different type for C's size_t than it does for std::size_t and ignore the warning.

closed 2020-03-11 by pabigot