zephyrproject-rtos/zephyr #21679: SPI broken on stm32f412 on master

StefJar, 2020-01-03:
I pulled the latest master and the SPI stopped working.

My config:
```
CONFIG_SPI=y
CONFIG_SPI_1=y
CONFIG_SPI_1_OP_MODES=1
CONFIG_SPI_3=y
CONFIG_SPI_3_OP_MODES=1
CONFIG_SPI_STM32_INTERRUPT=y
```
but it does not matter if we are using interrupts or not.

the code:
```
static sst26vf016b_error_t sst26vf016b_wakeup(void) {
	LOG_DBG("wakeup");
	u8_t buffer_tx[] = {sst26vf016b_op_RDPD};
	u8_t buffer_rx[1];
	struct spi_buf tx_buf [] = {
		{
		.buf = buffer_tx,
		.len = sizeof(buffer_tx)/sizeof(u8_t)
		}
	};
	const struct spi_buf_set tx = {
		.buffers = tx_buf,
		.count = sizeof(tx_buf)/sizeof(struct spi_buf)
	};
	const struct spi_buf rx_buf []= {
		{
		.buf = NULL,
		.len = 4,
		},
		{
		.buf = buffer_rx,
		.len = sizeof(buffer_rx),
		}
	};
	const struct spi_buf_set rx = {
		.buffers = rx_buf,
		.count = sizeof(rx_buf)/sizeof(struct spi_buf)
	};

	if (spi_transceive(sst26vf016b.spiDev, &sst26vf016b.spi_conf, &tx, &rx)) {
		return sst26vf016b_error_SPI;
	}

	if (SST26VF016B_JEDEC_DeviceID != buffer_rx[0]) {
		LOG_ERR("wrong device id! current DeviceID=0x%X vs DeviceID=0x%X",
				buffer_rx[0],
				SST26VF016B_JEDEC_DeviceID
			);
		return sst26vf016b_error_wakeup_wrongid;
	}
	k_sleep(K_MSEC(10)); // 10 msec till the device has left deep sleep
	return sst26vf016b_error_none;
}

```

My log print:
```
[00:00:01.421,000] <dbg> spi_ll_stm32.spi_stm32_configure: Installed config 0x20018ba4: freq 12000000Hz (div = 2), mode 0/0/0, slave 0
[00:00:01.434,000] <dbg> spi_ll_stm32.spi_context_buffers_setup: tx_bufs 0x200179a8 - rx_bufs 0x20017990 - 1
[00:00:01.444,000] <dbg> spi_ll_stm32.spi_context_buffers_setup: current_tx 0x200179b0 (1), current_rx 0x20017998 (2), tx buf/len 0x200179bc/1, rx buf/len 0x00000000/4
[00:00:01.459,000] <dbg> spi_ll_stm32.spi_context_update_tx: tx buf/len 0x00000000/0
```

on the logic analyzer:
[image]
[image]

Some comment:
1.) I checked the SPI coms via an older firmware that only uses STM32HAL and it works there. So we can say that the (flash) IC is not broken.
2.) there is no timeout, so the firmware breaks at https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/spi/spi_ll_stm32.c#L495 while waiting
3.) looks for me that the tx is send, but it somehow messes up while rx. There were some changes and maybe the `rx_buf` setup is no invaild

StefJar, 2020-01-03:
I forgot the old version for comparison:

`*** Booting Zephyr OS build zephyr-v2.1.0-383-g17d066b9dc0e  ***`

the latest version:

`*** Booting Zephyr OS build zephyr-v2.1.0-634-gbd9962d8d98c  ***`

And a print out for the old SPI com:

[image]

Marco-Peter, 2020-01-05:
I experienced the same issue with the effect that the SPI_NOR driver did not initialize anymore (it receives the wrong device id).
By trial and error I could locate the offending commit being 2ce8fa1e42d3d60e4119f15e82c29d2daaf67c00 but I still need to understand how.

StefJar, 2020-01-06:
@erwango I put your changes in but it is still broken
I attached a logic analser shot in.

old working version:
[image]

latest fix:
[image]

for me it looks that the rx never stops (spi_context_wait_for_completion does not return). That is why the CS does not go high

PR #21722 (2020-01-07): Revert "spi: stm32: Enable sending 16bits at a time in 8bit mode if p…
…ossible"

This reverts commit 2ce8fa1e42d3d60e4119f15e82c29d2daaf67c00.

Fixes #21546 
Fixes #21679

Following merge of https://github.com/zephyrproject-rtos/zephyr/pull/20948, following errors have been reported:
* https://github.com/zephyrproject-rtos/zephyr/issues/21546
* https://github.com/zephyrproject-rtos/zephyr/issues/21679

We've tried several changes and fixes on the change (cf #21712)
- Set the optimized behavior optional (default n)
- 3 additional fixes on the shift frame function flow 

Though issues remain in [... cut]
commit ca61847a (2020-01-07): Revert "spi: stm32: Enable sending 16bits at a time in 8bit mode if possible"

This reverts commit 2ce8fa1e42d3d60e4119f15e82c29d2daaf67c00.

Fixes #21546
Fixes #21679
closed 2020-01-07 by jhedberg