zephyrproject-rtos/zephyr #21681: nucleo_g431rb / STM32G4: Flashing works only once

martinjaeger, 2020-01-03:
**Describe the bug**
Flashing the board with the blinky or hello world sample as described in [Zephyr documentation](https://docs.zephyrproject.org/latest/boards/arm/nucleo_g431rb/doc/index.html#programming-and-debugging) works only once. Afterwards, PyOCD fails with the error as pasted below.

If I flash a very simple app built using other framework (e.g. Arduino Blinky) via the USB mass-storage method, I can afterwards flash with `west flash` again.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b nucleo_g431rb samples/basic/blinky/ --pristine`
2. `west flash`
3. `west flash` again
4. See error

**Expected behavior**
Flashing should also work with Zephyr app present on device.

**Impact**
Showstopper, as on my custom board I don't have the workaround of flashing via USB mass-storage device.

Debugging with Segger J-Link using Segger Ozone does not work either.

**Screenshots or console output**
```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /<private-path>/zephyr/build/zephyr/zephyr.hex
0000621:CRITICAL:__main__:STLink error (20): DP wait
Traceback (most recent call last):
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 344, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/__main__.py", line 474, in do_flash
    with session:
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 291, in __enter__
    self.open()
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 371, in open
    self._board.init()
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 83, in init
    self.target.init()
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 162, in init
    seq.invoke()
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 255, in power_up_debug
    self.write_reg(DP_SELECT, 0)
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 251, in write_reg
    self.write_dp(addr, data)
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 379, in write_dp
    self.probe.write_dp(addr, data)
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/probe/stlink_probe.py", line 164, in write_dp
    result = self._link.write_dap_register(STLink.DP_PORT, addr, data)
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/probe/stlink/stlink.py", line 425, in write_dap_register
    self._check_status(response)
  File "/home/martin/.local/lib/python3.8/site-packages/pyocd/probe/stlink/stlink.py", line 292, in _check_status
    raise self._ERROR_CLASSES[status](error_message)
pyocd.core.exceptions.TransferTimeoutError: STLink error (20): DP wait
ERROR: command exited with status 1: pyocd flash -e sector -a 0x8000000 -t stm32g431rbtx /<private-path>/zephyr/build/zephyr/zephyr.hex
```

**Environment (please complete the following information):**
 - OS: Linux 5.4.2-1-MANJARO #1 SMP PREEMPT Thu Dec 5 09:55:57 UTC 2019 x86_64 GNU/Linux
 - Toolchain: West v0.6.3, Zephyr SDK v0.10.3
 - Commit SHA: ac09bb33cfa03171f23683c43f14ede074d19469

**Additional context**
It seems like the problem could be caused by the MCU being in sleep mode, so that the debugger cannot connect to it anymore, see also [here](https://github.com/mbedmicro/pyOCD/issues/714).

If that's the case, does it make sense to set `DBGMCU_CR` register from somewhere within Zephyr application? I tried to enable `CONFIG_DEBUG=y` already, but it didn't make a difference.

FRASTM, 2020-01-06:
In the stm32g4/soc.c, the CONFIG_DEBUG flag can enable the DEBUG mode during sleep, stop and standby with :
> LL_DBGMCU_EnableDBGSleepMode();
> LL_DBGMCU_EnableDBGStopMode();
> LL_DBGMCU_EnableDBGStandbyMode();

Then, successive west flashing are possible. 

Note that another _manual_ solution is to Press the reset button and maintain shortly before launching the 'west flash' command

martinjaeger, 2020-01-06:
Awesome, that did the trick. Thanks for the quick response.

So how should we proceed? I could create a PR with the suggested changes. And we should include a remark in the board documentation that CONFIG_DEBUG=y has to be set in order to work properly.

Which other Nucleo boards could be affected by the same issue? Should we update all of them at once?

FRASTM, 2020-01-06:
There is also a user_script for the pyOCD which could write the DBGMCU_CR register on reset for STM32G4xx devices : [https://github.com/mbedmicro/pyOCD/blob/master/docs/user_scripts.md](url)
But I did not yet succeed with this method

def did_reset():
    # Set  SLEEP bit  to 1.
    target.write32(0xE0042004, 0x1)

erwango, 2020-01-20:
@martinjaeger, @FRASTM , I agree the pyocd script should be the best solution, since we don't want to impact binary for flashing purpose.
Using user script for flashing with pyocd is not available yet, but this is an on going work: https://github.com/zephyrproject-rtos/zephyr/pull/20957

Can you have a try with this?

martinjaeger, 2020-01-20:
I think it would still be good to disable the sleep mode if the debug flag is set (or add a separate Kconfig flag for this). Otherwise it's not possible to use e.g. Segger tools (J-Link and Ozone).

erwango, 2020-01-20:
I'm fine with the Kconfig flag proposal

PR #22496 (2020-02-05): soc: arm: stm32g4 disable sleep mode
This patch allows successive reflashing operation
on nucleo_g431rb boards

Fixes #21681
closed 2020-02-05 by galak