zephyrproject-rtos/zephyr #21707: Timing violation for all sensor drivers

sslupsky, 2020-01-05:
**Describe the bug**
I have been working with the BME280, SHT3X, OPT3001, ICM20648 drivers and I have noticed that there is no check after a power on event that the power on timing specification for the sensor is observed.  Most (all) sensors have a timing specification related to power on and accessing the device before the power on period has elapsed could lead to undefined behaviour.

**Impact**
Undefined behaviour of attached sensors.

I propose that the init function for all the sensors should be adapted to include a timeout check.  For example:

```
static int bme280_chip_init(struct device *dev)
{
	struct bme280_data *data = (struct bme280_data *) dev->driver_data;
	int err;
        s64_t now;

        now = k_uptime_get() * 1000;
        if (now < BME280_STARTUP_TIME_USEC ) {
            k_busy_wait(BME280_STARTUP_TIME_USEC - now);
        }
```

jhedberg, 2020-01-07:
@sslupsky since you had a fix proposal, could you perhaps submit a PR?

sslupsky, 2020-01-07:
I could do that ... might take a while because there are a lot of sensors in the tree and every one of them is affected by this.

I suppose the first thing is to discuss is if the proposed method is the way to proceed forward on this.

Also, should the start up time be specified in the DTS and configurable in KConfig?  If we did things that way, then the code could be generic.

jhedberg, 2020-01-07:
@sslupsky those are all fair questions. I'm not an expert in any of these areas but e.g. using DTS to describe the sensor timing requirements sounds reasonable to me. I suppose this issue may need to act as a "meta issue" and then sub-issues should be filed for the individual drivers?

@galak do you know who else should be added to the discussion for the things @sslupsky points out in his comment above?

jfischer-phytec-iot, 2020-01-07:
> I could do that ... might take a while because there are a lot of sensors in the tree and every one of them is affected by this.
>
> I suppose the first thing is to discuss is if the proposed method is [...]

To specify it in DTS is only necessary if the driver supports different devices with different power-on/start-up behavior. Also it depends on the device if k_busy_wait() is really necessary or k_sleep() is a better choice. Nothing more, no need to be more "generic". If there is a issue with the start-up time in BME280 driver and you can fix it, patches are welcome.

sslupsky, 2020-01-07:
@jfischer-phytec-iot My point is all the sensors in the tree are affected by this issue.  So, the amount of work to go fix all of them is quite significant.  If we use a generic piece of code for this issue and put the timing parameter in the DTS, then we could probably push the need to set the specific timing requirement to the DTS and could use some "default" value(s).  The user can easily override the default value in the DTS without having the change the driver.  Over a longer period of time, the default values would be properly identified and coded.  I do not know if this really accomplishes much versus having the values hard coded in the device driver and doing them one by one.  But, I think I generally like the idea of fixing this for all devices at once since it is one of those edge cases that can cause someone's head to explode trying to debug it.

You are correct on the k_busy_wait() versus k_sleep().  Though, at the time the drivers are loading, are the threads running yet?  That is, I do not believe you can switch context at pre_kernel so during this phase it is irrelevant.  Post_kernel you probably can switch context?  Thus, k_sleep() might be an option here provided there is no driver dependency that requires a driver to be loaded before another one.  If that is the case, then some sort of API would be required so that a driver could tell if it had to wait for another one to load first.
 
This is just my line of thinking at the time I mentioned it.

closed 2020-08-28 by github-actions[bot]