zephyrproject-rtos/zephyr #21713: CDC ACM USB class issue with high transfer rate

binkboyd, 2020-01-06:
**Describe the bug**
This issue is possibly linked to the following two issues: https://github.com/zephyrproject-rtos/zephyr/issues/20987 and https://github.com/zephyrproject-rtos/zephyr/issues/20250

Using an NRF52840  when sending data from the device to host  above 10kb/s continuously  - After a short time the console will show error messages of the form of:
<pre>
<wrn> usb_device: Failed to write endpoint buffer 0x82
<wrn> usb_device: Failed to write endpoint buffer 0x82
</pre>
The speed needed varies and it seems also to be related to the state of the host USB stack, as a restart of host sometimes removes the symptoms for a small period of time. When the issue is triggered data is lost, and counting bytes sent on the devices vs bytes received on the host you can see a mismatch with fewer bytes received on the host than sent from the device. 

Using the same fix as in https://github.com/zephyrproject-rtos/zephyr/issues/20250 (disable ZLP) the issue disappears and data is transferred correctly without any warnings in the log.
<pre>
diff --git a/subsys/usb/class/cdc_acm.c b/subsys/usb/class/cdc_acm.c
index 3f7082374d..0a3bcf25a4 100644
--- a/subsys/usb/class/cdc_acm.c
+++ b/subsys/usb/class/cdc_acm.c
@@ -303,7 +303,7 @@ static void tx_work_handler(struct k_work *work)
        LOG_DBG("Got %d bytes from ringbuffer send to ep %x", len, ep);
-       usb_transfer(ep, dev_data->tx_buf, len, USB_TRANS_WRITE,
+       usb_transfer(ep, dev_data->tx_buf, len, USB_TRANS_WRITE | USB_TRANS_NO_ZLP,
                     cdc_acm_write_cb, dev_data);
 }
</pre>

**To Reproduce**
Steps to reproduce the behavior:
1. Build the CDC-ACM example for an NRF52840 board (samples/subsys/usb/cdc_acm)
2. Send incrementing data as fast as you are allowed and watch the log output for failed to write endpoint buffer warnings.
3. Count increment on receiving end and see gaps in the transferred data

**Expected behavior**
All data sent from device should be received on the host

**Impact**
Data loss is basically a showstopper for our use, as this functions as a bridge interface for incoming data over a different link.

**Environment (please complete the following information):**
- OS: Linux (Ubuntu 18.04) 
- Toolchain: Zephyr SDK 0.10.3
- Commit: Latest zephyr master (currently https://github.com/zephyrproject-rtos/zephyr/commit/85a08da966e3a2c4e9d3012bcfd4e54c24b6003f for cdc_acm.c)

**Additional context**
Maybe it's worth investigating if there is something wonky with the ZLP handling in the USB stack as this is the second class where disabling ZLP appears to fix an issue.

jfischer-phytec-iot, 2020-01-06:
@binkboyd How do you read the data on the host side? Could you provide an example, please?

sslupsky, 2020-01-06:
@binkboyd Please see my recent comment on #20987 

Related to this, I understand ZLP's are essential to the USB spec.  So, it seems counter intuitive to me to suppress them.  I think the issue may be related to "dropping" the packet when the USB hardware is still in the process of sending the packet.  The packet isn't actually dropped though, as it is retried again and almost always goes through the second time.

binkboyd, 2020-01-06:
@jfischer-phytec-iot  I'm using pyserial to receive 32k blocks on the host side. I'll try to create a minimal test case when I get some time. 

@sslupsky That's possible, I haven't debugged this in depth. I tried to increase the retry count but if I recall correctly it crashed after a while if the retry count was increased past the default value of 3. I can see if I can condense all this into a test case along with a pyserial script to read from the host side.

sslupsky, 2020-01-06:
@binkboyd In one of my earlier tests, if the USB hardware is busy, waiting 500us (1/2 the USB slot time) eliminated the warnings as well.  I suspect the usb driver is attempting to send a ZLP while a transfer is ongoing.

jfischer-phytec-iot, 2020-01-07:
Some thought,

From USB2.0 Spec, 5.8.3 Bulk Transfer Packet Size Constraints
```
An endpoint must always transmit data payloads with a data field less than or equal to the endpoint’s
reported wMaxPacketSize value. When a bulk IRP involves more data than can fit in one maximum-sized
data payload, all data payloads are required to be maximum size except for the last data payload, which will
contain the remaining data. A bulk transfer is complete when the endpoint does one of the following:
• Has transferred exactly the amount of data expected
• Transfers a packet with a payload size less than wMaxPacketSize or transfers a zero-length packet
When a bulk transfer is complete, the Host Controller retires the current IRP and advances to the next IRP.
If a data payload is received that is larger than expected, all pending bulk IRPs for that endpoint will be
aborted/retired.
```
IRP is I/O Request Packets (on the host side). We will always have this ZLP issues on different places because how it was defined. How I understand, the length of the transfer buffer (acm->readsize) in Linux kernel ACM driver is mostly 2 * endpoint buffer size or less. But on the device side we can not know the length of the request, so probably transfer of a ZLP is not really wrong :-) on the other side if it transfers exactly the amount of data or less than MPS the ZLP is not necessary. Like in btusb the transfer callback triggers an other transfer but I do see what leads the host to stop IRP after a ZLP.

emob-nordic, 2020-01-21:
I am currently trying to reproduce this issue but I am failing to do so. Do You have any minimal example which triggers described issue so I could go further into investigation?

jfischer-phytec-iot, 2020-01-27:
@carlescufi @emob-nordic We tended to patch it and insert USB_TRANS_NO_ZLP flag, I experimented a little with pyserial. Scenario like in the #22236 does not work with USB_TRANS_NO_ZLP. I think it needs more investigation.

carlescufi, 2020-02-05:
Summary of root cause:

- The Zephyr application sends bytes via the UART APIs, which then get encoded into USB packets. If the packet size at some point is exactly equal to the max packet size (MPS, 64 bytes) and ZLP is enabled, the Zephyr USB stack will send a ZLP. Then the Host or the application running on the Host may believe that there is no more data to be received (i.e. the transaction has completed), hence not triggering another IN transaction, and thus preventing any more data to be sent from Zephyr to the Host.

- If ZLP is **not** enabled, then sending a packet which is exactly MPS (64 bytes) in length should not make the application on the Host believe that there is no more data to be received. That said, this depends on the application on the Host and how it is written.

** Zephyr Application Workaround**

- Always send MPS-1 or fewer bytes at a time (i.e. for every call to a `uart_*` API)

carlescufi, 2020-02-05:
Additional context:

- DAPLink also sends ZLP in the same way as Zephyr, see [here](https://github.com/ARMmbed/DAPLink/blob/master/source/usb/cdc/usbd_cdc_acm.c#L666)
- Tests run with Segger J-Link show that at least under normal conditions, packets sent by it are always smaller than MPS. Additional testing is needed to confirm.

sslupsky, 2020-02-05:
Doesn't what you described conflict with the USB specification?  A ZLP should only be sent when the transaction is complete?  If it is not complete, the USB stack should keep sending MSP's until complete and then a ZLP at the end.  Or, if the last packet is not MSP, then it just sends the last packet.

sslupsky, 2020-02-05:
I actually think this problem is not related to ZLP's.  I think there is a problem with some of the drivers.  From what I could see in the Nordic and sam0 drivers, the drivers do not wait for the hardware if the hardware is busy sending a packet, the drivers drop the packet.

I fixed this in the sam0 driver and I haven't seen this problem since.  Though, I have not done any formal testing or stress testing to confirm that.

jfischer-phytec-iot, 2020-02-05:
> Doesn't what you described conflict with the USB specification? A ZLP should only be sent when the transaction is complete? If it is not complete, the USB stack should keep sending MSP's until complete and then a ZLP at the end. Or, if the last packet is not MSP, then it just sends the last packet.

ZLP should be sent when a transfer is complete and last packet is equal to MPS (64 bytes for FS device). For the CDC ACM sample, if one fills the buffer with 64 bytes at once then Zephyr's ACM class implementation would trigger a transfer of 64 bytes + ZLP. This behavior is valid from stacks POV. To avoid this behavior, do not fill UART buffer with data length equal to a multiple of endpoint MPS.

jfischer-phytec-iot, 2020-02-05:
> I actually think this problem is not related to ZLP's. I think there is a problem with some of the drivers. From what I could see in the Nordic and sam0 drivers, the drivers do not wait for the [...]

You probably did not consider usb_transfer and usb_transfer_callback.

sslupsky, 2020-02-05:
I took a deep dive into the USB transfer mechanics a while ago.  There is a busy mechanism that works well to avoid software bottlenecking.  Where it breaks is here (sam0 driver):
https://github.com/zephyrproject-rtos/zephyr/blob/e3d35713f608f676667c9012b7a076aa5f78f562/drivers/usb/device/usb_dc_sam0.c#L547-L550

This checks the status of the hardware buffer.   If it is busy, it drops the packet and sends you back to usb_write(), which generates the endpoint buffer warning.

I looked at the Nordic driver and I see the same issue.  When the hardware is busy, it is possible the packet will be dropped.  The probability of a dropped packet increases with the amount of data and time between calls to the usb_write() function.

I see a couple ways to address this:

1. Do a busy wait in the driver and wait for the hardware buffer to be available.  
2. Revisit the transfer mechanism and introduce a hardware status check.

sslupsky, 2020-02-05:
> ZLP should be sent when a transfer is complete and last packet is equal to MPS (64 bytes for FS device). For the CDC ACM sample, if one fills the buffer with 64 bytes at once then Zephyr's ACM class [...]

Exactly my understanding of how this works.  Not sure why you would want to "avoid" this behaviour?

jfischer-phytec-iot, 2020-02-05:
> This checks the status of the hardware buffer. If it is busy, it drops the packet and sends you back to usb_write(), which generates the endpoint buffer warning.

This behavior is correct and usb_transfer_work would not call usb_write() until the driver triggers usb_transfer_ep_callback().

> I looked at the Nordic driver and I see the same issue. When the hardware is busy, it is possible the packet will be dropped. The probability of a dropped packet increases with the amount of data [...]

(1) We must not busy wait there. (2) We do that in usb_transfer.

nvlsianpu, 2020-02-19:
I observed the issue along with mcuboot recovery via usb cdc acm transmission. On 50% firmware update try I have got failure due the failure to write endpoint buffer 0x82. I used configuration I posted here https://github.com/JuulLabs-OSS/mcuboot/pull/665 on nrf52840_pca10056 board.

jfischer-phytec-iot, 2020-02-19:
Thanks for report @nvlsianpu

closed 2020-02-19 by jhedberg

tmon-nordic, 2023-10-26:
> I actually think this problem is not related to ZLP's. I think there is a problem with some of the drivers. From what I could see in the Nordic and sam0 drivers, the drivers do not wait for the hardware if the hardware is busy sending a packet, the drivers drop the packet.

Not necessarily drivers, but I don't quite like the way `usb_transfer_work()` works:
https://github.com/zephyrproject-rtos/zephyr/blob/9b851b2ab813b0bca588f0afeff7d5e878c0e789/subsys/usb/device/usb_transfer.c#L87-L95

Which means that if we have to transmit ZLP, then the stack will arm the IN endpoint with ZLP **and** will set `trans->status` to 0 and thus effectively call transfer completion callback. This opens a race condition between the host and firmware (device running Zephyr). If Zephyr wins the race, then driver will drop the next scheduled packet because it is still waiting for host to read the ZLP. If host wins the race, then the issue is not really visible.

PR #22798 (2020-02-19): usb: cdc_acm: TX memory and performance improvements. Avoid ZLP transfer
Currently the cdc_acm implementation send data to the USB stack by
chunks of USB MPS. This has 2 drawbacks:
- when more data than USB MPS has to be sent, each chunk needs 2 packets
  due to the need of an extra ZLP;
- a temporary buffer of size USB MPS is needed.

This patch improves the memory consumption and performances by passing
the ring buffer directly to the USB stack. It only has a small
performance degradation when the ring buffer wraps and less than USB MPS
has to be sent.

Fixes #21713.
commit 28b7953e (2020-02-19): usb: cdc_acm: do not send ZLP for IN transactions

When the length of the transfer is an exact multiple of IN endpoint MPS
in the ring buffer, transfer one byte less to avoid zero-length packet.
Otherwise the application running on the host may conclude that there
is no more data to be received (i.e. the transaction has completed),
hence not triggering another I/O Request Packet (IRP).

Fixes #21713.