zephyrproject-rtos/zephyr #21715: nucleo_g431rb: Blinky too slow / wrong clock setup?

martinjaeger, 2020-01-06:
**Describe the bug**
The LED is on approx. 15 times per minute (measured with stop watch), so it is 2 times slower than expected. On = off time in blinky app is set to 1 second.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_g431rb samples/basic/blinky/
2. west flash

**Impact**
As I'm dealing with energy measurement in my application, the energy will most probably be wrong by the same factor as the blinky interval is wrong.

**Environment:**
- OS: Linux 5.4.2-1-MANJARO #1 SMP PREEMPT Thu Dec 5 09:55:57 UTC 2019 x86_64 GNU/Linux
- Toolchain: West v0.6.3, Zephyr SDK v0.10.3
- Commit SHA: ac09bb3

**Additional context**
I double-checked the clock and PLL configuration in `nucleo_g431rb_defconfig` vs. ST reference manual already and could not find any mistake.

FRASTM, 2020-01-07:
This is due to a AHBPrescaler not correct in the LL_PLL_ConfigSystemClock_HSI() function

martinjaeger, 2020-01-07:
Great that you found the reason. So will you provide a fix?

FRASTM, 2020-01-07:
Refer to stm32cube: stm32g431: wrong clock setup #34

martinjaeger, 2020-01-07:
> Refer to stm32cube: stm32g431: wrong clock setup #34

Don't understand what exactly you are referring to. Was that a commit / PR? Number 34 doesn't seem related to our problem.

FRASTM, 2020-01-07:
Sorry, refer to Refer to stm32cube: stm32g431: wrong clock setup zephyrproject-rtos/hal_stm32#34

closed 2020-01-10 by jhedberg