zephyrproject-rtos/zephyr #21716: nucleo_g431rb: Hello world not working

martinjaeger, 2020-01-06:
**Describe the bug**
The basic hello world example does not work for the Nucleo G431RB (STM32G431 MCU).

**To Reproduce**
Try the steps according to [board description in docs](https://docs.zephyrproject.org/latest/boards/arm/nucleo_g431rb/doc/index.html) and see that hello world does not output anything on the terminal.

**Environment (please complete the following information):**
- OS: Linux 5.4.2-1-MANJARO #1 SMP PREEMPT Thu Dec 5 09:55:57 UTC 2019 x86_64 GNU/Linux
- Toolchain: West v0.6.3, Zephyr SDK v0.10.3
- Commit SHA: ac09bb3

**Additional context**
The [Nucleo board manual](https://www.st.com/resource/en/user_manual/dm00556337.pdf) states on page 24: "Communication between the target STM32G4 and the STLINK-V3E MCU is enabled on LPUART1 to support the Virtual COM port". 

However, the board DTS file selects UART1 for the console: https://github.com/zephyrproject-rtos/zephyr/blob/5308a941a9670e64b3a019ae4ce65f984a915ce7/boards/arm/nucleo_g431rb/nucleo_g431rb.dts#L16

Just changing it to use `&lpuart1` didn't fix the problem. As this issue might be related to the clock configuration (#21715), I also tried with baud rates divided by two on host side without any success.

LPUART on the board should be connected to PA2 (TX) and PA3 (RX), so that's correct in board spec. However, looking at [pinmux.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nucleo_g431rb/pinmux.c) file, `PA2` is used for both `USART2` and `LPUART1`, which might not be the best idea. I wonder if there was any reason for that or if it was just a mistake.

azeemshatp, 2020-01-06:
There seems to be another mistake in the doc in that case (https://docs.zephyrproject.org/latest/boards/arm/nucleo_g431rb/doc/index.html). 
It says the Zephyr console output is assigned to UART2. 
Does this mean DTS and doc are not matching?.

FRASTM, 2020-01-07:
Actually, the nucleo_g431rb board outputs the console on LPUART1 (on VCP_Tx/RX pin PA2, PA3)
This has to be changed in the board config (and doc. ). I will proceed to update that.

Note that baudrate > 115200 is reached with LPUART1 clock source different from LSE clock.

martinjaeger, 2020-01-07:
Please have a look at the PR mentioned above. I guess that implements what you describe.

I tested the fix and it worked fine with my Nucleo board.

FRASTM, 2020-01-07:
That's fine, we could also choose to output USART2 on PA2, PA3  (and leave LPUART on PC1, PC0 if needed). 
It works if USART2 is selected for console.
The nucleo_g431rb board refers to UART1 and LPUART1, should it be mentioned USART2,  USART3 ( available on pin PB10, PB11 ) and UART4  (available on pin PC10, PC11).

martinjaeger, 2020-01-07:
Well, you're right. We could wire UART2 to those pins. But in the board manual they state that LPUART is the default for the VCP (as far as I understand it). So I guess it makes sense to keep it that way to avoid confusion and be consistent with the manual.

So you are saying that we should also configure UART3 and 4? Not sure if that's needed by anyone, but would be easy to add to the PR if you like to have it.

FRASTM, 2020-01-07:
I agree, we keep LPUART 1 for console. However for UART_2 pin PA14 & PA15 are already assigned to T_SWCLK and T_JDTI,  so could we omit any reference to UART2 in  `boards/arm/nucleo_g431rb/pinmux.c`  for nucleo_g431rb.
--> only UART1 and LPUART1 are mentioned.

An interesting note ( in the `index.rst`) is  that _LPUART1 baudrate is limited to 9600 if it is clocked by LSE._

martinjaeger, 2020-01-07:
Ok, I'll update the PR accordingly.

PR #21739 (2020-01-13): boards: nucleo_g431rb: Default serial port fixed
The virtual COM port of the STlink in the Nucleo board is connected to
LPUART1, but the board was configured to use UART1 instead. For this
reason, hello world sample did not work.

In addition to that, PA2 was assigned to both LPUART1 and UART2. UART2
TX is now muxed to PA14.

Fixes #21716.
closed 2020-01-13 by ioannisg