zephyrproject-rtos/zephyr #21724: dts: edtlib: handle child-binding or child-child-binding as 'normal' binding with compatible

b0661, 2020-01-06:
**Is your enhancement proposal related to a problem? Please describe.**

I want to define the bindings for the pinctrl state nodes of pin controllers and the pin configuration nodes of pin controllers as child-binding within the pin-controller binding (see additional context). Edtlib does not recognize the compatible in a child-binding or a child-child-binding.

**Describe the solution you'd like**

Handle child-bindings the same way as 'normal' bindings. Allow at least for one level of child-binding
within a child-binding.

Edtlib should find three compatibles ("level_0", "level_1", "level_2) and three bindings in the following binding file example:

```
description: Level 0

compatible: "level_0"

child-binding:
    description: Level 1

    compatible: "level_1"

    child-binding:
        description: Level 2

        compatible: "level_2"
```

**Describe alternatives you've considered**

Define the "pinctrl-state" binding and the "pincfg" binding in separate binding files.

**Additional context**

The pin controller binding looks like this:

```
# Copyright (c) 2018..2020 Bobby Noelte
# SPDX-License-Identifier: Apache-2.0

description: Pin controller

include: base.yaml

properties:
    pin-controller:
      type: boolean
      required: true
      description: device controller identification

    "#pinctrl-cells":
      type: int
      required: false
      description: >
        Number of pin control cells in addition to the index within the
        pin controller device instance

child-binding:
    description: >
      Pinctrl state. A node that bundles a set of pin configuration
      nodes that shall be applied in a single pin control operation.

    compatible: "pinctrl-state"

    child-binding:
        description: Pin configuration

        compatible: "pincfg"

        groups:
          descripion: >
            The list of names of the groups that properties in the node apply to
            (either this, "pins", "pinmux" or "pinctrl-pin-array" have to be
            specified).
          required: false
          type: string-array

        pins:
          descripion: >
            The list of pin identifiers that properties in the node apply to
            (either this, "groups", "pinmux" or "pinctrl-pin-array" have to be
            specified).
            The binding for the specific pin controller hardware defines whether
            the entries are of type int-array or string-array.
          required: false

        function:
          required: false
          type: string

        pinmux:
          descripion: >
            The list of numeric pin ids and their mux settings that properties
            in the node apply to (either this, "pins", "groups" or
            "pinctrl-pin-array" have to be specified).
          required: false
          type: int-array

        pinctrl-pin-array:
          descripion: >
            The list of pin controller register index and values (either this,
            "pins", "pinmux" or "groups" have to be specified). pinctrl-cells
            specifies the number of value cells in addition to the index of the
            registers. No other properties shall be in the node.
          required: false
          type: int-array

        bias-disable:
          description: Disable any pin bias.
          required: false
          type: boolean

        bias-high-impedance:
          required: false
          type: boolean

        bias-bus-hold:
          required: false
          type: boolean

        bias-pull-up:
          description: Pull up strength in Ohm. 0 to disable pull up.
          required: false
          type: int

        bias-pull-down:
          description: Pull down strength in Ohm. 0 to disable pull down.
          required: false
          type: int

        bias-pull-pin-default:
          required: false
          type: int

        drive-push-pull:
           required: false
           type: boolean

        drive-open-drain:
           required: false
           type: boolean

        drive-open-source:
            required: false
            type: boolean

        drive-strength:
           description: >
             Sink or source at most X mA. 0 to disable drive strength control.
           required: false
           type: int

        input-enable:
            required: false
            type: boolean

        input-debounce:
            description: Debounce time in usec. 0 to disable debouncing.
            required: false
            type: int

        input-disable:
            required: false
            type: boolean

        input-schmitt-enable:
            required: false
            type: boolean

        input-schmitt-disable:
            required: false
            type: boolean

        low-power-enable:
            description: Enable low power mode.
            required: false
            type: boolean

        low-power-disable:
            description: Disable low power mode.
            required: false
            type: boolean

        output-disable:
            required: false
            type: boolean

        output-enable:
            required: false
            type: boolean

        output-low:
            required: false
            type: boolean

        output-high:
            required: false
            type: boolean

        power-source:
            description: Numerical ID of power source to select.
            required: false
            type: int

        slew-rate:
            description: Slew rate
            required: false
            type: int

        skew-delay:
            description: >
              Expected clock skew on input pins and the delay before latching
              a value to an output pin.
              Typically indicated by the number of double-inverters used to
              delay the signal.
            required: false
            type: int
```

The pinctrl-state and pincfg bindings are currently attached to the respective nodes by a patched version of edtlib.Node._init_binding.

```
# @brief Initialize Node.matching_compat, Node._binding, and Node.binding_path
#
# Specially treat the /chosen node, /aliases node and the pin controller
# pinctrl state and pin configuration nodes to attach a default binding.
#
# Monkey patch for edtlib.Node._init_binding.
# To be used as long as the extended functionality is not part of edtlib.
#
# @param self refers to edtlib.Node object
def _edtlib_Node_init_binding(self):
    # Initializes Node.matching_compat, Node._binding, and
    # Node.binding_path.
    #
    # Node._binding holds the data from the node's binding file, in the
    # format returned by PyYAML (plain Python lists, dicts, etc.), or None
    # if the node has no binding.

    # This relies on the parent of the node having already been
    # initialized, which is guaranteed by going through the nodes in
    # node_iter() order.
    print("Monkey patch '{}' called."
          .format(inspect.currentframe().f_code.co_name))

    if "compatible" in self._node.props:
        self.compats = self._node.props["compatible"].to_strings()
    elif self._node.name == "chosen":
        # use default compatible for chosen node
        self.compats = ["chosen"]
    elif self._node.name == "aliases":
        # use default compatible for aliases node
        self.compats = ["aliases"]
    elif self._node.parent and \
         self._node.parent.props.get("pin-controller", False):
        # child of pin controller node
        if self._node.nodes:
            # the pin controller sub nodes has sub nodes in itself
            # Assume it is a pinctrl state node
            self.compats = ["pinctrl-state"]
        else:
            # No sub nodes - assume a pin configuration node
            self.compats = ["pincfg"]
    elif self._node.parent and self._node.parent.parent and \
         self._node.parent.parent.props.get("pin-controller", False):
        # Grand child of pin controller node
        # Assume  a pin configuration node
        self.compats = ["pincfg"]
    else:
        self.compats = []

    if self.compats:
        on_bus = self.on_bus

        for compat in self.compats:
            if (compat, on_bus) in self.edt._compat2binding:
                # Binding found
                self.matching_compat = compat
                self._binding, self.binding_path = \
                    self.edt._compat2binding[compat, on_bus]
                return
    else:
        # No 'compatible' property. See if the parent binding has a
        # 'child-binding:' key that gives the binding (or a legacy
        # 'sub-node:' key).
        binding_from_parent = self._binding_from_parent()
        if binding_from_parent:
            self._binding = binding_from_parent
            self.binding_path = self.parent.binding_path
            self.matching_compat = self.parent.matching_compat
            return

    # No binding found
    self._binding = self.binding_path = self.matching_compat = None
```

ulfalizer, 2020-01-07:
What should happen if some node has `compatible = "level-1"` but isn't the child of a node with `compatible = "level-0"`? Seems a bit confusing to allow a compatible in a mechanism that says "give this binding to child nodes".

I'll look at it some more later, but I wonder if there might be a simpler way.

ulfalizer, 2020-01-07:
Is this all about being able to give a custom `Node.matching_compat` to nodes that get their binding from `child-binding`? Might be able to work something out in that case.

b0661, 2020-01-07:
> Is this all about being able to give a custom Node.matching_compat to nodes that get their binding from child-binding?

The evident problem in my use case is that  **_init_compat2binding()** does not extract child bindings from a binding file. Therefor the child bindings are also __NOT__ associated to a compatible if there is any in the child binding.  

In a broader scope the problem may be that there is no mechanism to associate a binding to a node that is a child-node or child-child-node of a node that has a binding with recursive child-bindings.

I don't use a general mechanism for that in the pinctrl use case because the child-node relation may be
  1) pin-controller -> pincfg **OR**
  2) pin-controller -> pinctrl-state -> pincfg.

These pinctrl child-node relation variants seem also an issue in #20398 and may make a general mechanism more complex.

The pin-controller binding is an approximation of child-node relation type 2 (the one "somehow" preferred by Linux). The specific binding to node association does some heuristics and therefor also covers the type 1 child-node relation.  

```
    if "compatible" in self._node.props:
        self.compats = self._node.props["compatible"].to_strings()
    ...
    elif self._node.parent and \
         self._node.parent.props.get("pin-controller", False):
        # child of pin controller node
        if self._node.nodes:
            # the pin controller sub nodes has sub nodes in itself
            # Assume it is a pinctr state node
            self.compats = ["pinctrl-state"]
        else:
            # No sub nodes - assume a pin configuration node
            self.compats = ["pincfg"]
    elif self._node.parent and self._node.parent.parent and \
         self._node.parent.parent.props.get("pin-controller", False):
        # Grand child of pin controller node
        # Assume  a pin configuration node
        self.compats = ["pincfg"]
    else:
        self.compats = []
```

> Seems a bit confusing to allow a compatible in a mechanism that says "give this binding to child nodes".

I though this is the most simple, take the compatible mechanism that is well known.

> What should happen if some node has compatible = "level-1" but isn't the child of a node with compatible = "level-0"?

Nothing, just take the compatible as it is given. I would not construct a deep DTS error analysis for such a specific case.

mbolivar-nordic, 2022-10-12:
I stumbled across this issue while looking at something else.

> In a broader scope the problem may be that there is no mechanism to associate a binding to a node that is a child-node or child-child-node of a node that has a binding with recursive child-bindings.

We have this now. Multiple levels of child binding have been supported for a while. I'm therefore going to close this issue as completed. If I've missed something and today's edtlib is still missing a feature, please reopen. Thanks.

closed 2022-10-12 by mbolivar-nordic