zephyrproject-rtos/zephyr #21735: ARM: Cortex-M: IRQ lock/unlock() API accessible but non-functional from user mode

ioannisg, 2020-01-07:
**Describe the bug**

IRQ API is a kernel API, so, normally it shall not be accessible from user mode. This is the case for most of the API functions; an attempt to use them leads to a CPU fault.

This is **not** the case, however, for `irq_enable()`, `irq_disable()` for ARM Cortex-M Mainline architecture (both ARMv7-M and ARMv8-M): these functions do not cause a fault if called from user mode, although, they (correctly) do not lock or unlock interrupts. Since we do not explicitly document that the IRQ API shall only be used from supervisor mode, this might end up very confusing for user threads calling irq_lock/unlock API.

**To Reproduce**
Steps to reproduce the behavior:
Simply call irq_lock() from user mode on ARM Cortex-M Mainline. The call does not fail and does not lock the IRQs.

**Expected behavior**
It should be clear for user mode threads that irq lock /unlock cannot be used.

I would see two option here
- either add an assert on irq_lock, unlock when called from user mode
- or simply document explicitly that the API shall not be used from user mode, otherwise we might have undefined behavior.

ioannisg, 2020-01-07:
CC @andrewboie

jhedberg, 2020-01-07:
Discussed in the release meeting: The documentation update needs to be done either way, however an assert is probably nice too since otherwise this risks hard-to-catch bugs with critical regions seemingly working but actually not.

pderwin, 2020-01-07:
I'll echo the 'hard-to-catch bugs' commentary.  I have been actively pursuing a problem on Cortex-R in user mode that turned out to be this exact issue.  Our user-mode code believed it was locking a critical section, but in reality was not.  

Now investigating how to re-factor the code to rely on semaphore instead of interrupt disable.

PR #21752 (2020-01-08): irq: add note about irq_lock/unlock access
On ARM irq_lock() simply fails silently instead of generating
an exception.

Fixes: #21735
commit 06df06b7 (2020-01-08): irq: add note about irq_lock/unlock access

On ARM irq_lock() simply fails silently instead of generating
an exception.

Fixes: #21735
closed 2020-01-08 by nashif