zephyrproject-rtos/zephyr #21745: tests: counter_basic_api: Failed on stm32 based boards

erwango, 2020-01-07:
**Describe the bug**
Test failed on following boards: nucleo_l476rg, nucleo_wb55rg

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
Test should pass.

**Screenshots or console output**
```
Running test suite test_counter
===================================================================
starting test - test_set_top_value_with_alarm
Skipped for RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_set_top_value_with_alarm
===================================================================
starting test - test_single_shot_alarm_notop
Testing RTC_0

    Assertion failed at ../../src/test_counter.c:297: test_single_shot_alarm_instance: (0 not equa)
RTC_0: Setting top value to default failed
FAIL - test_single_shot_alarm_notop
===================================================================
starting test - test_single_shot_alarm_top
Skipped for RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_single_shot_alarm_top
===================================================================
starting test - test_multiple_alarms
Skipped for RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_multiple_alarms
===================================================================
starting test - test_all_channels
Testing RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_all_channels
===================================================================
starting test - test_late_alarm
Skipped for RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_late_alarm
===================================================================
starting test - test_late_alarm_error
Skipped for RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_late_alarm_error
===================================================================
starting test - test_short_relative_alarm
Testing RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_short_relative_alarm
===================================================================
starting test - test_cancelled_alarm_does_not_expire
Skipped for RTC_0

    Assertion failed at ../../src/test_counter.c:102: counter_tear_down_instance: (0 not equal to )
RTC_0: Setting top value to default failed
FAIL - test_cancelled_alarm_does_not_expire
===================================================================
Test suite test_counter failed.
===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

jhedberg, 2020-01-07:
@erwango will you be looking at this? If not, please reassign.

erwango, 2020-01-24:
In `counter_tear_down_instance`, `counter_set_top_value(dev, &top_cfg)` is called with following `top_cfg` value:
```
	struct counter_top_cfg top_cfg = {
		.callback = NULL,
		.user_data = NULL,
		.flags = 0
	};
```
Then, it fails on error.

I don't think this call is matching description of  `counter_set_top_value` contract in counter.h.
@nordic-krch as counter API owner, can you verify?

PR #22262 (2020-01-28): tests: drivers: counter: counter_basic_api: Fix teardown function
Teardown may fail due to not supported setting of top_value.
Modified to accept not supported case.

Partially fixes #21745 

It fixes error caught in teardown function.
PR #22305 (2020-01-31): tests: drivers: counter: counter_basic_api: Fix single_shot_alarm
test_single_shot_alarm_instance may fail due to not supported setting
of top_value in call to counter_set_top_value.
Modified to accept not supported case.

Fixes #21745
commit 41a3ac6a (2020-01-31): tests: drivers: counter: counter_basic_api: Fix single_shot_alarm

test_single_shot_alarm_instance may fail due to not supported setting
of top_value in call to counter_set_top_value.
Modified to accept not supported case.

Fixes #21745
closed 2020-01-28 by nashif