zephyrproject-rtos/zephyr #21754: Arduino Due shell does not accept input (UART0)

adamfeuer, 2020-01-08:
**Describe the bug**

I am trying to get a shell running on the Arduino Due. I can compile and flash [`samples/subsys/shell/shell_module`](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/shell/shell_module) on my Arduino Due. I am using Ubuntu Linux 19.10..

Connected to the Due's Programming Port using picocom, I can see output and the shell prompt (`uart:~$`). However, when I try to type something, the shell appears hung– it doesn't accept any input. It appears the UART is not receiving anything.

Is this sample or UART0 input in general known to work on the Arduino Due?

I can output characters on the UART0 just fine using printk(). I tried to do some printk() debugging– the Arduino seems to know it's getting characters, but the actual character is not supplied to the shell.

When I send a character to the Arduino Due using picocom, the UART0 interrupt fires. But when I try to read from the UART0 UART_RHR, the shell gets 0, not the character I send. And if I check the uart->UART_SR, it shows UART_SR_FRAME – a framing error.

I know I can build working zephyr software since I built a posix version of the shell examples and they work.

The Arduino Due board can read characters just fine from using Arduino software, if I flash an Arduino sketch; or from NuttX RTOS if I build that and flash it onto the board. So the hardware is good. It seems to be something related to how Zephyr configures the board. I've been reading the uart init code and board init code to see what's wrong with it, but can't find a problem.

I have a Segger J-Link debugger and cable I can use with the Arduino Due, but not sure what to look for.

Can you help?

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p auto -b arduino_due samples/subsys/shell/shell_module`
1. `west flash`
1. `picocom -b 115200 /dev/ttyACM0`
1. Zephyr prompt appears. Try to type commands– there is no response.

**Expected behavior**
I expect to be able to type `<RETURN>` and at least get another prompt; or to type `help<RETURN>` and get the zephyr shell help text.

**Impact**
Shell is unusable, so I can't use to compile and run other samples that use the shell; or develop and debug applications that use the shell; or easily debug problems with TCP/IP networking.

**Screenshots or console output**
```
*** Booting Zephyr OS build zephyr-v2.1.0-668-g4eee43e420ba  ***

uart:~$
```
**Environment (please complete the following information):**
 - OS: Ubuntu Linux 19.10
 - Toolchain: Zephyr SDK, installed from current master
 - Commit SHA or Version used: `7d2586533d5f27bb7fbe50fb03cc50a4dfe30077`

**Additional context**
None

sven-hm, 2020-03-27:
I ran into the same problem but could fix it with a `10k` pull-up between `RX0` and `3.3V`.

ioannisg, 2020-09-10:
@adamfeuer could you check the suggestion by @sven-hm and report if this could fix your issue?

closed 2020-09-14 by MaureenHelm

adamfeuer, 2020-09-15:
@ioannisg Why should I have to modify my board to use the shell? I don't have to do that on NuttX.

sven-hm, 2020-09-15:
@adamfeuer i think @ioannisg refers to #27317. There the internal pull-up is activated for the arduino_due board, so no need to modify the hardware.
By the way, NuttX had [the same problem](https://github.com/apache/incubator-nuttx/commit/4868acf27e699b9af4d8bedf55be56d36d97b8a5) and fixed it by activating the internal pull-up as well.

adamfeuer, 2020-09-15:
@sven-hm @ioannisg Thanks for explaining, this makes sense. I will try the new code.

PR #27317 (2020-09-14): boards: arduino_due: Enable internal pullup on uart rx line
This PR enables the internal pull-up resistor on the uart rx line to the onboard serial chip. This fixes #21754.

Additional information:
[Compare ArduinoCore-sam library](https://github.com/arduino/ArduinoCore-sam/commit/89caaafa20398560a6f13e75749fbe6b0addb15e)