zephyrproject-rtos/zephyr #21755: tests/drivers/adc/adc_api  failed on  mec15xxevb_assy6853 board.

chen-png, 2020-01-08:
**To Reproduce**
Steps to reproduce the behavior:

1.  programming the tests/drivers/adc/adc_api to mec15xxevb_assy6853 board, according to
https://docs.zephyrproject.org/latest/boards/arm/mec1501modular_assy6885/doc/index.html to set up.
2. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.1.0-744-g7ee5cf6a0bd1 ***

Running test suite adc_basic_test
starting test - test_adc_sample_one_channel
Samples read: 0x0000
**Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[0] should be non-zero**

FAIL - test_adc_sample_one_channel
starting test - test_adc_sample_two_channels
Samples read: 0x0000
Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[0] should be non-zero

FAIL - test_adc_sample_two_channels
starting test - test_adc_asynchronous_call
Samples read: 0x0000
Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[0] should be non-zero

FAIL - test_adc_asynchronous_call
starting test - test_adc_sample_with_interval
sample_with_interval_callback: sampling 0
sample_with_interval_callback: sampling 1
sample_with_interval_callback: sampling 2
sample_with_interval_callback: sampling 3
sample_with_interval_callback: sampling 4
Samples read: 0x0000
Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[0] should be non-zero

FAIL - test_adc_sample_with_interval
starting test - test_adc_repeated_samplings
repeated_samplings_callback: done 1
Samples read: 0x0000
Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[0] should be non-zero
0x0000
Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[1] should be non-zero
0x0000 0x0000 0x0000 0x0000
repeated_samplings_callback: done 2
Samples read: 0x0000
Assertion failed at ZEPHYR_BASE/tests/drivers/adc/adc_api/src/test_adc.c:243: check_samples: (0 equal to sample_value)
[0] should be non-zero
ASSERTION FAIL [!(thread->base.user_options & ((1UL << (0))))] @ ZEPHYR_BASE/arch/arm/core/aarch32/cortex_m/thread_abort.c:36
essential thread aborted
E: r0/a1: 0x00000004 r1/a2: 0x00000024 r2/a3: 0x84000000
E: r3/a4: 0x000e807c r12/ip: 0x00000000 r14/lr: 0x000e1b9d
E: xpsr: 0x6100005e
E: Faulting instruction address (r15/pc): 0x000e606e
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x00118790 (main)
E: Halting system

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: g7ee5cf6a0b

ashaannajoy, 2020-02-14:
Comment from dcpleung: requires specific hardware setup for ADC to work (jumpers and a reasonable voltage source). It possibly needs pinmux changes in code. IIRC, the ADC pins are being used for other purposes in the EC code where this test would not work as-is

dcpleung, 2020-02-20:
On header J35, instead of connecting pins 1-2 and 4-5, try pin 1-2 and 3-4. This provides the ADC pins with ~2.048V with Vref at 3.0V. With 12-bit ADC, this should give you values around 0xAF0.

225296, 2020-02-27:
After changing the jumper on header **J35 to pin 1-2 and 3-4** , Sanity check is **PASSED**

Environment (please complete the following information):
OS : Ubuntu LTS 18.04
Toolchain: Zephyr SDK : 0.11.2
Commit ID: g7b4861114372

closed 2020-03-02 by dcpleung

chen-png, 2020-03-20:
> After changing the jumper on header **J35 to pin 1-2 and 3-4** , Sanity check is **PASSED**

@225296  could you show me how to connect pins on JP35? 
 I don't have the board manual for mec15xxevb_assy6853, and i found that there are totally 3 pins in JP35.

225296, 2020-03-20:
@chen-png  JP35 is not the header we use for ADC ,
On header **J35**, has totally 5 pins instead of connecting pins 1-2 and 4-5, try pin **1-2 and 3-4** on mec15xxevb_assy6853 . It should **PASS**

Environment (please complete the following information):
OS : Ubuntu LTS 18.04
Toolchain: Zephyr SDK : 0.11.2
Commit ID: g7b4861114372

chen-png, 2020-03-23:
@225296  thanks, i will have a try.

chen-png, 2020-04-03:
@225296 i shorted pin 1-2 and pin 3-4 and ran this test, it wouldn't report any error, but it would stop at "test_adc_repeated_samplings", do you know how to fix this?

*** Booting Zephyr OS build zephyr-v2.2.0-1091-gc98652b42f12  ***
Running test suite adc_basic_test

starting test - test_adc_sample_one_channel
Samples read: 0x0af1 0x0000 0x0000 0x0000 0x0000 0x0000 
PASS - test_adc_sample_one_channel

starting test - test_adc_sample_two_channels
Samples read: 0x0af2 0x0af5 0x0000 0x0000 0x0000 0x0000 
PASS - test_adc_sample_two_channels

starting test - test_adc_asynchronous_call
Samples read: 0x0af2 0x0af8 0x0af8 0x0af8 0x0af8 0x0000 
PASS - test_adc_asynchronous_call

starting test - test_adc_sample_with_interval
sample_with_interval_callback: sampling 0
sample_with_interval_callback: sampling 1
sample_with_interval_callback: sampling 2
sample_with_interval_callback: sampling 3
sample_with_interval_callback: sampling 4
Samples read: 0x0af8 0x0af9 0x0af2 0x0af0 0x0af0 0x0000 
PASS - test_adc_sample_with_interval

**starting test - test_adc_repeated_samplings**
repeated_samplings_callback: done 1
Samples read: 0x0af2 0x0af6 0x0000 0x0000 0x0000 0x0000 
repeated_samplings_callback: done 2
Samples read: 0x0af2 0x0af6 0x0af8 0x0af6 0x0000 0x0000 
repeated_samplings_callback: done 3
Samples read: 0x0af2 0x0af6 0x0af2 0x0af5 0x0000 0x0000 
repeated_samplings_callback: done 4
Samples read: 0x0af2 0x0af6 0x0af9 0x0af6 0x0000 0x0000

dcpleung, 2020-04-03:
This is due to the board going into deep sleep and not waking up.

chen-png, 2020-04-07:
how to fix this? otherwise, it will always be judged as failed test cases by sanity check script.

dcpleung, 2020-04-07:
In current power management scheme, it is possible to turn on DEVICE_POWER_MANAGEMENT and modify the device driver to take advantage of this. It is also possible to modify the residency so the test would not fail.