zephyrproject-rtos/zephyr #21756: tests/kernel/obj_tracing failed on mec15xxevb_assy6853 board.

chen-png, 2020-01-08:
**To Reproduce**
Steps to reproduce the behavior:

1.  programming the tests/kernel/obj_tracing  to mec15xxevb_assy6853 board, according to
https://docs.zephyrproject.org/latest/boards/arm/mec1501modular_assy6885/doc/index.html to set up.
2. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.1.0-744-g7ee5cf6a0bd1 ***

Running test suite obj_tracing
starting test - test_philosophers_tracing
SEMAPHORE REF: 0x00118628
SEMAPHORE REF: 0x00118610
SEMAPHORE REF: 0x001185f8
SEMAPHORE REF: 0x001185e0
SEMAPHORE REF: 0x001185c8
SEMAPHORE REF: 0x00118ba4
SEMAPHORE REF: 0x00118104
SEMAPHORE REF: 0x00118cc0
SEMAPHORE REF: 0x00118cdc
SEMAPHORE REF: 0x0011cd74
SEMAPHORE REF: 0x0011cd5c
SEMAPHORE QUANTITY: 5
COOP: 0x00118640 OPTIONS: 0x00, STATE: 0x40
COOP: 0x0011893c OPTIONS: 0x00, STATE: 0x02
COOP: 0x001188a4 OPTIONS: 0x00, STATE: 0x02
COOP: 0x0011880c OPTIONS: 0x00, STATE: 0x40
COOP: 0x00118774 OPTIONS: 0x00, STATE: 0x02
COOP: 0x001186dc OPTIONS: 0x00, STATE: 0x40
PREMPT: 0x00118eac OPTIONS: 0x00, STATE: 0x02
COOP: 0x00118124 OPTIONS: 0x00, STATE: 0x02
COOP: 0x00118cf4 OPTIONS: 0x01, STATE: 0x00
COOP: 0x00118d8c OPTIONS: 0x01, STATE: 0x40
THREAD QUANTITY: 10

**Assertion failed at ZEPHYR_BASE/tests/kernel/obj_tracing/src/main.c:120: object_monitor: (((thread_counter == TOTAL_THREADS) && (obj_counter == TOTAL_OBJECTS)) is false)
test failed**

FAIL - test_philosophers_tracing
starting test - test_obj_tracing

PASS - test_obj_tracing
Test suite obj_tracing failed.
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: g7ee5cf6a0b

nashif, 2020-01-08:
I am sure this is not a bug, looks like the board is enabling all kind of options that most likely have some kernel objects, so the counting does not match.

nashif, 2020-01-20:
this is not a bug, the PS2 and other IPs being enabled by default add objects and the count does not match anymore, so the test fails. Boards should by default only enable the bare minimum, all optional features should only be enabled by the application using them, and not by default for every configuration 

cc @albertofloyd @franciscomunoz

franciscomunoz, 2020-02-14:
The current number of threads "10" did not matched the expected "9". The kscan driver has a thread which is messing with the expected count. The test passes after disabling it ... I'm going to create a PR for disabling the KSCAN in the board.

starting test - test_philosophers_tracing
SEMAPHORE REF: 0x00118088
SEMAPHORE REF: 0x00118070
SEMAPHORE REF: 0x00118058
SEMAPHORE REF: 0x00118040
SEMAPHORE REF: 0x00118028
SEMAPHORE REF: 0x00118604
SEMAPHORE REF: 0x00118b50
SEMAPHORE REF: 0x00118b6c
SEMAPHORE REF: 0x00118730
SEMAPHORE REF: 0x00118718
SEMAPHORE REF: 0x0011cbf8
SEMAPHORE REF: 0x0011cbe0
SEMAPHORE QUANTITY: 5
COOP: 0x001180a0 OPTIONS: 0x00, STATE: 0x80
COOP: 0x0011839c OPTIONS: 0x00, STATE: 0x02
COOP: 0x00118304 OPTIONS: 0x00, STATE: 0x02
COOP: 0x0011826c OPTIONS: 0x00, STATE: 0x80
COOP: 0x001181d4 OPTIONS: 0x00, STATE: 0x02
COOP: 0x0011813c OPTIONS: 0x00, STATE: 0x80
PREMPT: 0x00118d3c OPTIONS: 0x00, STATE: 0x02
COOP: 0x00118b84 OPTIONS: 0x01, STATE: 0x00
COOP: 0x00118c1c OPTIONS: 0x01, STATE: 0x80
THREAD QUANTITY: 9
PASS - test_philosophers_tracing
===================================================================
starting test - test_obj_tracing
PASS - test_obj_tracing
===================================================================
Test suite obj_tracing succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

https://github.com/zephyrproject-rtos/zephyr/pull/22816

ashaannajoy, 2020-02-14:
This test passes by disabling KSCAN in zephyr/boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig

nashif, 2020-02-19:
#22921 has a more generic fix that should work with pre-existing conditions. Please verify.

PR #22921 (2020-02-19): test: object_tracing: account for pre-existing threads
Count existing threads before the test has started to deal with
platforms that have some existing services.

Fixes #21756
commit 28298fa0 (2020-02-19): test: object_tracing: account for pre-existing threads

Count existing threads before the test has started to deal with
platforms that have some existing services.

Remove hard-coded accounting for IPM, this is now counted before the
test starts.

Fixes #21756
PR #26962 (2020-09-04): [Backport v1.14-branch] test: object_tracing: account for pre-existing threads
Count existing threads before the test has started to deal with
platforms that have some existing services.

Remove hard-coded accounting for IPM, this is now counted before the
test starts.

Fixes #21756

Fixes #26595
Backport of #22921
commit 4da6a5bc (2020-09-04): test: object_tracing: account for pre-existing threads

Count existing threads before the test has started to deal with
platforms that have some existing services.

Remove hard-coded accounting for IPM, this is now counted before the
test starts.

Fixes #21756
closed 2020-02-19 by nashif