zephyrproject-rtos/zephyr #21762: [v1.14] stm32: k_sleep() actual sleep times are different than its input

gon1332, 2020-01-08:
**Describe the bug**
`k_sleep()` generally sleeps more than the intentional time. Starting from 1 millisecond and for exactly every 10 values there is a new fixed time that the system sleeps. I've used the Analog Discovery 2 to contact the measurements on a GPIO pin. The effect seems consistent as you can see in the graph I've made: 

[image]

[Here you can see](https://docs.google.com/spreadsheets/d/1YQMMPMakEMSU_ecJ7lDWIILajGtjWGfmRYB9No_ZpkY/edit?usp=sharing) the spreadsheet with the measurements.

This is obvious for small sleep times but, it is not so observable for greater sleep times.

I haven't tried to diagnose the issue. However, on master (f5d2b3271a), the behavior is the expected one.

**To Reproduce**

I've tried the below code on nucleo_l496zg on top of v1.14-branch (3132a41600348a6a546c55c35c8f46822973c459):

```
#include <zephyr.h>
#include <device.h>
#include <gpio.h>

#define PIN_CONTROLLER	"GPIOF"
#define PIN	13

struct device *gpiof;

void test_sleep(int msec)
{
    gpio_pin_write(gpiof, PIN, 1);
    k_sleep(msec);
    gpio_pin_write(gpiof, PIN, 0);
}

void main(void)
{
    gpiof = device_get_binding(PIN_CONTROLLER);
    gpio_pin_configure(gpiof, PIN, GPIO_DIR_OUT);
    gpio_pin_write(gpiof, PIN, 0);

    /* You can change the range as you please */
    for (u16_t i = 1; i <= 20; i++) {
        test_sleep(i);
        k_busy_wait(100);
    }
}
```

**Expected behavior**
I expected the sleep times that I put as input to `k_sleep()`.

**Impact**
Really weird and frustrating behaviors on the final application.

**Environment (please complete the following information):**
 - OS: GNU/Linux
 - Toolchain Zephyr SDK

gon1332, 2020-01-08:
@galazari

erwango, 2020-01-15:
@jhedberg, STM32 platform are ticked by cortex_m_systick. Only impact on 'k_sleep' that could be seen from stm32 code could be the configuration which would have a constant impact.

Reported issue is dynamic, so I'd say that potential issue lays either in 'k_sleep' or cortex_m_systick timer code. I propose to check fisrt cortex_m_systick.

@ioannisg, can you provide your view on recent cortex_m_systick evolutions that could have had a positive impact on reported issue?

Maybe a kernel expert would have a better view on `k_sleep` evolution between 1.14 and recent master ?

nashif, 2020-09-16:
close as stale

closed 2020-09-16 by nashif