zephyrproject-rtos/zephyr #21772: Adding I2C devices to device tree with the same address on different busses generates excessive warnings.

jbroxson-fbrl, 2020-01-08:
**Describe the bug**
Given the following scenario, build warnings are generated when adding devices to device tree:

* I2C1:
  * Device address: 0x48
  * Device address: 0x49
* I2C2:
  * Device address: 0x48

**To Reproduce**
Add 3 devices to the device tree as shown below. Two devices on different busses must have identical I2C addresses:

```
&i2c1 {
	status = "okay";
	max31875_0: max31875@48 {
		compatible = "maxim,max31875";
		reg = <0x48>;
		status = "okay";
		label = "MAX31875_0";
	};
	max31875_1: max31875@49 {
		compatible = "maxim,max31875";
		reg = <0x49>;
		status = "okay";
		label = "MAX31875_1";
	};
};
&i2c2 {
	status = "okay";
	max31875_2: max31875@48 {
		compatible = "maxim,max31875";
		reg = <0x48>;
		status = "okay";
		label = "MAX31875_2";
	};
};
```
Building will generate the warnings shown in **Screenshots or console output** below. Making any of the following changes will cause the build warnings to go away (but are not acceptable workarounds):

* Disable any of the three devices
* Change  `reg` for `max31875_2` to `0x50` (or any non-conflicting value)

**Results**

**Expected behavior**
To be able to add devices with the same I2C address to different busses without build errors or warnings.

**Impact**
We consider this a blocking issue for our board bring-up. The amount of warnings generated is severe and makes it difficult to discern other warnings and errors.

**Screenshots or console output**
Build warnings for one file being built. These warning repeat multiple times.
```
[161/162] Building C object zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj
In file included from /home/djbroxson/xxx/third-party/zephyr/zephyr_rtos/v2.1.0/zephyr/include/generated_dts_board.h:16:0,
                 from /home/djbroxson/xxx/third-party/zephyr/zephyr_rtos/v2.1.0/zephyr/include/arch/arm/arch.h:20,
                 from /home/djbroxson/xxx/third-party/zephyr/zephyr_rtos/v2.1.0/zephyr/include/arch/cpu.h:17,
                 from zephyr/isr_tables.c:7:
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:338:0: warning: "DT_MAXIM_MAX31875_0_BASE_ADDRESS" redefined
 #define DT_MAXIM_MAX31875_0_BASE_ADDRESS __DEPRECATED_MACRO   DT_ST_STM32_I2C_V2_40005800_MAXIM_MAX31875_48_BASE_ADDRESS
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:233:0: note: this is the location of the previous definition
 #define DT_MAXIM_MAX31875_0_BASE_ADDRESS __DEPRECATED_MACRO   DT_ST_STM32_I2C_V2_40005400_MAXIM_MAX31875_48_BASE_ADDRESS
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:339:0: warning: "DT_MAXIM_MAX31875_0_BUS_NAME" redefined
 #define DT_MAXIM_MAX31875_0_BUS_NAME __DEPRECATED_MACRO   DT_ST_STM32_I2C_V2_40005800_MAXIM_MAX31875_48_BUS_NAME
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:234:0: note: this is the location of the previous definition
 #define DT_MAXIM_MAX31875_0_BUS_NAME __DEPRECATED_MACRO   DT_ST_STM32_I2C_V2_40005400_MAXIM_MAX31875_48_BUS_NAME
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:340:0: warning: "DT_MAXIM_MAX31875_0_LABEL" redefined
 #define DT_MAXIM_MAX31875_0_LABEL __DEPRECATED_MACRO    DT_ST_STM32_I2C_V2_40005800_MAXIM_MAX31875_48_LABEL
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:235:0: note: this is the location of the previous definition
 #define DT_MAXIM_MAX31875_0_LABEL __DEPRECATED_MACRO    DT_ST_STM32_I2C_V2_40005400_MAXIM_MAX31875_48_LABEL
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:341:0: warning: "DT_MAXIM_MAX31875_0_STATUS" redefined
 #define DT_MAXIM_MAX31875_0_STATUS __DEPRECATED_MACRO    DT_ST_STM32_I2C_V2_40005800_MAXIM_MAX31875_48_STATUS
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:236:0: note: this is the location of the previous definition
 #define DT_MAXIM_MAX31875_0_STATUS __DEPRECATED_MACRO    DT_ST_STM32_I2C_V2_40005400_MAXIM_MAX31875_48_STATUS
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:342:0: warning: "DT_MAXIM_MAX31875_0_STATUS_ENUM" redefined
 #define DT_MAXIM_MAX31875_0_STATUS_ENUM __DEPRECATED_MACRO   DT_ST_STM32_I2C_V2_40005800_MAXIM_MAX31875_48_STATUS_ENUM
zephyr/include/generated/generated_dts_board_unfixed.h.deprecated:237:0: note: this is the location of the previous definition
 #define DT_MAXIM_MAX31875_0_STATUS_ENUM __DEPRECATED_MACRO   DT_ST_STM32_I2C_V2_40005400_MAXIM_MAX31875_48_STATUS_ENUM
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK / GCC
 - Zephyr version: 2.1.0

**Additional context**
N/A

galak, 2020-01-14:
max31875 doesn't seem to be a driver in the code base.  As I'm assuming this is your own driver, then you are using deprecated DTS generated macro's and thus you are getting those warnings. This is expected behavior.

jbroxson-fbrl, 2020-01-16:
Yes - this is a custom driver. I apologize for my lack of knowledge on this one, but can you point me to something which shows avoiding "using deprecated DTS generated macro" to correct the issue?

Are you referring to our use of DEVICE_DEFINE() within the driver itself or something else within how we use DTS?

jbroxson-fbrl, 2020-01-23:
I did manage to create a workaround for this, however it's a bit of a sledgehammer.  Adding the following line to the end out our CMakeLists.txt file causes the .h.deprecated file to be overridden and thus eliminates the warnings.  We would however like to understand how to properly correct this issue if it is not an actual bug.

```
# Avoid issues described here: https://github.com/zephyrproject-rtos/zephyr/issues/21772
file(WRITE ${GENERATED_DTS_BOARD_UNFIXED_H}.deprecated "/* FBRL: This file is not used. */")
```

galak, 2020-07-21:
@jbroxson-fbrl you need to convert your driver to use the new macros.

closed 2020-07-21 by galak