zephyrproject-rtos/zephyr #21780: OpenThread fails on nRF52840 Dongle (nrf52840_pca10059)

braiden, 2020-01-09:
**Describe the bug**
Enabling OpenThread L2 on nrf52840_pca10059 doesn't work. Crashes with a bus error. 

`zephyr/subsys/net/lib/openthread/platform/flash.c` wants to use the last two pages of flash for OpenThread. On this board the stock boot-loader occupies the end of flash, and also configures the ACL peripheral to prevent flash writes to this address region. This causes a bus error when OpenThread tries to setup flash.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/sockets/echo_server
2. west build -b nrf52840_pca10059 -p auto -- -DCONF_FILE="prj.conf overlay-ot.conf"
3. nrfutil pkg generate --debug-mode --sd-req 0x00 --hw-version 52 --application build/zephyr/zephyr.hex zephyr.zip
4. nrfutil dfu usb-serial --serial-number F5FCB4F99C88 --package zephyr.zip
5. Fails with:

```
[00:00:00.000,000] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.006,530] <err> os: ***** BUS FAULT *****
[00:00:00.006,530] <err> os:   Imprecise data bus error
[00:00:00.006,561] <err> os: r0/a1:  0x000fe000  r1/a2:  0x00000000  r2/a3:  0x00000001
[00:00:00.006,561] <err> os: r3/a4:  0x4001e000 r12/ip:  0x00000000 r14/lr:  0x0000ae6b
[00:00:00.006,561] <err> os:  xpsr:  0x21000000
[00:00:00.006,591] <err> os: Faulting instruction address (r15/pc): 0x0000d690
[00:00:00.006,591] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.006,591] <err> os: Current thread: 0x20007450 (unknown)
[00:00:00.079,040] <err> os: Halting system
```

 0x0000d690 is within nrf nvm erase code.

~/opt/zephyr/sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-addr2line -e build/zephyr/zephyr.elf 0x0000d690
/home/braiden/opt/zephyr/zephyr/modules/hal/nordic/nrfx/drivers/src/nrfx_nvmc.c:223

**Expected behavior**
Add some KConfig for OpenThread flash partition or address?

**Impact**
Minor?

My work-around was to use "storage" flash partition configured in device tree:
Replace `ot_flash_offset = info.start_offset;` in `zephyr/subsys/net/lib/openthread/platform/flash.c` with `ot_flash_offset = FLASH_AREA_STORAGE_OFFSET`.

**Screenshots or console output**

See _To Reproduce_.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.10.3
 - Commit: tags/zephyr-v2.1.0

**Additional context**
Add any other context about the problem here.

rlubos, 2020-01-09:
It's a known issue, unfortunately, see https://github.com/zephyrproject-rtos/zephyr/issues/13941.

I have a solution on one of my branches, which introduces settings module for OpenThread based on Zepyr settings module. It has its flaws due to some shortcomings in the OpenThread/Zephyr API compatibility, it is a working solution though. I'll try to prepare a PR out of it, perhaps we could make it in.

rlubos, 2020-01-09:
btw, there is a one-year-old PR open, applying the workaround you've proposed, it got stuck though: https://github.com/zephyrproject-rtos/zephyr/pull/12944

rlubos, 2020-01-17:
Resolved by #21796.

closed 2020-01-17 by rlubos