zephyrproject-rtos/zephyr #21783: rename zassert functions

andrewboie, 2020-01-09:
Coverity thinks any function with the word "assert" in its name is a conditionally compiled expression, and generates false positives about side effects.

To avoid this hassle, rename the zassert... family of APIs used in our test cases to something which doesn't have "assert" in the same. zcheck perhaps?

The existing APIs should be marked with __deprecated and just call the new APIs. All test cases should be converted.

jukkar, 2020-01-17:
I am not familiar with Coverity so do not know if this is possible, but IMHO we should instead try to configure Coverity to not generate these false positives. It sounds wrong to change our code just to satisfy some weird rule in a tool.

nashif, 2021-05-28:
agree with @jukkar

closed 2021-05-28 by nashif