zephyrproject-rtos/zephyr #21789: Merge topic-gpio back to master

carlescufi, 2020-01-09:
In order to merge topic-gpio back to master we need to finalize:

- [ ] Complete #18530  
- [ ] Complete #20017
- Add missing items to release notes:
- [ ] pin enable/disable callback operations disabled

jfischer-phytec-iot, 2020-01-09:
Could we skip #20017 and convert GPIO users on master? That would avoid conflict and locks.

carlescufi, 2020-01-09:
> Could we skip #20017 and convert GPIO users on master? That would avoid conflict and locks.

As discussed before, this would cause sanitycheck to fail because in-tree samples would be using deprecated functions. If you want to propose an alternative mechanism to merge topic-gpio I suggest you take a look first at  @pabigot's [initial roadmap proposal](https://github.com/zephyrproject-rtos/zephyr/issues/18530#issuecomment-527997161).

pabigot, 2020-01-27:
This comment provides more detail on how I intend to address several of the main issue bullet points.  It may be adjusted based on discoveries and feedback.

Work will begin once #18530 is complete, and #20017 is nearly complete.  At that point the topic branch will again be rebased on master.

The exact order and breakdown of changes into into commits and PRs will be determined as I get into it.  Work-in-progress will probably be exposed in a draft PR.  My intent is to have all this submitted for review by 01 February so we can merge the topic branch to master the week of 03 February.  The deadline is 07 February, but there's no reason to wait until the last minute.

### Generic

* API documentation clarifications not covered by other updates.

### Role-based typedef cleanup

* Change public GPIO API to use `gpio_pin_t` for all parameters that are pin indexes.
* Replace `gpio_devicetree_flags_t` to `gpio_dt_flags_t`.
* Replace `u32_t` by `gpio_port_value_t` for all port API value parameters.

### Function and macro deprecation

* Re-implement `gpio_pin_write()` as `gpio_pin_set_raw()` and deprecate it. Replace all in-tree uses.
* Re-implement `gpio_pin_read()` as `gpio_pin_get_raw()` and deprecate it. Replace all in-tree uses.
* Macro deprecation details TBD.

### Implementation cleanup
* (all drivers) Remove the write and read API functions and references to `GPIO_ACCESS_BY_` macros.
* (all drivers) Remove access_op parameter from API functions.
* (all drivers) Update API function declarations to use role-specific typedefs
* Remove the write and read API functions and `GPIO_ACCESS_BY_` macros from the API function table and generic header.
* Fix return value for internal `get_pending_int` function declaration.

carlescufi, 2020-02-18:
Closing this since it's not relevant anymore

closed 2020-02-18 by carlescufi