zephyrproject-rtos/zephyr #21790: tests/kernel/timer/timer_api fails on nucleo_g071rc board

erwango, 2020-01-09:
**Describe the bug**
`test_timer_k_define` fails on nulco_g071rb

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Screenshots or console output**
```
*** Booting Zephyr OS build zephyr-v2.1.0-393-ge928d6d16e4f  ***                
Running test suite timer_api                                                    
===================================================================             
starting test - test_time_conversions                                           
PASS - test_time_conversions                                                    
===================================================================             
starting test - test_timer_duration_period                                      
PASS - test_timer_duration_period                                               
===================================================================             
starting test - test_timer_period_0                                             
PASS - test_timer_period_0                                                      
===================================================================             
starting test - test_timer_expirefn_null                                        
PASS - test_timer_expirefn_null                                                 
===================================================================             
starting test - test_timer_periodicity                                          
PASS - test_timer_periodicity                                                   
===================================================================             
starting test - test_timer_status_get                                           
PASS - test_timer_status_get                                                    
===================================================================             
starting test - test_timer_status_get_anytime                                   
PASS - test_timer_status_get_anytime                                            
===================================================================             
starting test - test_timer_status_sync                                          
PASS - test_timer_status_sync                                                   
===================================================================             
starting test - test_timer_k_define                                             
                                                                                
    Assertion failed at ../../src/main.c:64: duration_expire: interval >= 100 ie
                                                                                
FAIL - test_timer_k_define                                                      
===================================================================             
starting test - test_timer_user_data                                            
PASS - test_timer_user_data                                                     
===================================================================             
starting test - test_timer_remaining_get                                        
PASS - test_timer_remaining_get                                                 
===================================================================             
Test suite timer_api failed.                                                    
===================================================================             
PROJECT EXECUTION FAILED  
```

erwango, 2020-02-05:
Tested passed on 41a3ac6a35ea18dd70b4f056d205e75dce428e48

erwango, 2020-02-05:
An always enjoyable session of `git bisect` pointed me to 96ccc46e03058877b052945b25295aeb2a73da2d

One funny thing is that failing point has changed, but assert remain the same:
```
*** Booting Zephyr OS build zephyr-v2.1.0-1645-g96ccc46e0305  ***                                                                                
Running test suite timer_api                                                                                                                     
===================================================================                                                                              
starting test - test_time_conversions                                                                                                            
PASS - test_time_conversions                                                                                                                     
===================================================================                                                                              
starting test - test_timer_duration_period                                                                                                       
                                                                                                                                                 
    Assertion failed at ZEPHYR_BASE/tests/kernel/timer/timer_api/src/main.c:64: duration_expire: interval >= 100 is false
                                                                                                                                                 
FAIL - test_timer_duration_period                                                                                                                
===================================================================                                                                              
starting test - test_timer_period_0                                                                                                              
PASS - test_timer_period_0                                                                                                                       
===================================================================                                                                              
starting test - test_timer_expirefn_null                                                                                                         
PASS - test_timer_expirefn_null                                                                                                                  
===================================================================                                                                              
starting test - test_timer_periodicity                                                                                                           
PASS - test_timer_periodicity                                                                                                                    
===================================================================                                                                              
starting test - test_timer_status_get                                                                                                            
PASS - test_timer_status_get                                                                                                                     
===================================================================                                                                              
starting test - test_timer_status_get_anytime                                                                                                    
PASS - test_timer_status_get_anytime                                                                                                             
===================================================================                                                                              
starting test - test_timer_status_sync                                                                                                           
PASS - test_timer_status_sync                                                                                                                    
===================================================================                                                                              
starting test - test_timer_k_define                                                                                                              
PASS - test_timer_k_define                                                                                                                       
===================================================================                                                                              
starting test - test_timer_user_data                                                                                                             
PASS - test_timer_user_data                                                                                                                      
===================================================================                                                                              
[... cut ...]
===================================================================                                                                              
Test suite timer_api failed.                                                                                                                     
===================================================================                                                                              
PROJECT EXECUTION FAILED  
```

@andyross, can you have a check?
@ioannisg maybe you have a clue?

andyross, 2020-02-05:
Will take a look.  That change should be identical modulo some irq_lock timing though, which (on what I think is a 64MHz cpu!) makes it sort of a confusing candidate for a failure in a subsystem where timing resolution is 6400 cycles.

Is this a reliable failure or spurious?  Does it happen on all STM32 boards or just this particular one?  Unfortunately I don't have any STM boards, but I can see about getting one.

andyross, 2020-02-05:
Oh, and you confirmed that reverting just that one patch in isolation on top of master resolves the issue?  Or is this just the bisection endpoint?

andyross, 2020-02-05:
Unfortunately I've had no success reproducing this on the hardware I have available.  The SysTick devices I have locally are the sam_e70_xplained and the frdm_k64f, and both pass.  They're both rather faster CPUs than the STM32 though.  So I dug out a nrf52840_pca10056, which is also a 64 MHz board.  It natively uses its own timer device but can be made to run with SysTick identically to your board using:

```
CONFIG_NRF_RTC_TIMER=n
CONFIG_CORTEX_M_SYSTICK=y
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=64000000
CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000
```

And that runs correctly too (obviously the nRF board also runs fine with its own timer).

Is it possible there's a timing configuration mixup on this particular board that just happens to have been unmasked?  Maybe the listed frequency in our configuration doesn't match the value it actually gets at boot or something?

erwango, 2020-02-06:
@andyross , thanks for feedback. Here are some inputs to your questions:

> Is this a reliable failure or spurious? Does it happen on all STM32 boards or just this particular one? Unfortunately I don't have any STM boards, but I can see about getting one.

100% reproducible.
This board is the only impacted one AFAIK.

> Oh, and you confirmed that reverting just that one patch in isolation on top of master resolves the issue?
Reverting the guilty commit makes it pass on current master.

erwango, 2020-02-06:
Testing on current master (7c95d503b0a64810ace17f058cf0178c18acc816), test is passed.

Commit getting test passed again is : 7c95d503b0a64810ace17f058cf0178c18acc816
Which is not at all related with timer (but at least impact the binary in some marginal way).

Building with `west build -b nucleo_g071rb tests/kernel/timer/timer_api -c --pristine `
Tested 10 times on previous commit, test is 100% failed (twice in a row, plugging, unplugging, ...)
Tested 10 times on 7c95d503b0a64810ace17f058cf0178c18acc816, test is 100% passed (twice in a row, plugging, unplugging)
Tested on previous commit + guilty commit revert, test is passed (100%)

I can't explain this fact, but at least test is passed.
I propose to close the issue, unless someone wants to dig into impact of `z_mrsh_gpio_pin_interrupt_configure` assembly code variation on timer_api testing.

andyross, 2020-02-06:
OK.  FWIW: if I had to make a wild guess, there's probably some board level-misconfiguration at work here.  Maybe something like the DTS file lists 64 MHz as per the docs, but the bootloader/firmware/ROM/whatever actually configures the PLL at 32 MHz by default instead, etc..

closed 2020-02-07 by erwango

erwango, 2020-02-07:
> FWIW: if I had to make a wild guess, there's probably some board level-misconfiguration at work here. Maybe something like the DTS file lists 64 MHz as per the docs, but the bootloader/firmware/ROM/whatever actually configures the PLL at 32 MHz by default instead, etc..

Thanks @andyross  I'll have a look.