zephyrproject-rtos/zephyr #21798: Bluetooth: host:

wdhpawa, 2020-01-09:
**Is your enhancement proposal related to a problem? Please describe.**
When a GATT client subscribes to notifications from a server, the client host layer stores some subscription info in bt_gatt_subscribe_params structs (and possibly elsewhere depending on PR #21682). If the client loses this subscription info (e.g. due to a reboot), there is currently no way to restore it without resubscribing (i.e. resending the subscription requests over the air to the server). In the case of bonded devices, the server will preserve subscriptions (BT Core spec v5.1 vol 3 part G section 3.3.3.3), so in this case resubscribing is a waste of time and power.

**Describe the solution you'd like**
The host should provide a way to restore the subscription info on the client side without resubscribing. One solution might be to persist this info via the settings subsystem, but this might be unnecessary as most of the info (handles, subscription values, and peer addresses) would normally already be known and persisted (or hardcoded in the application logic) by the application layer or host in the context of service discovery and bonding, and the remaining info (node, ...) should be fairly easy for the host to recreate. So it would be preferable if the host would provide an API allowing the application layer to collaborate with the host to restore the subscription info - without the need for resubscribing or persisting duplicate info.

**Describe alternatives you've considered**
\-

**Additional context**
This issue was originally part of issue  #21103.

carlescufi, 2020-01-14:
@Vudentz or @joerchan any feedback?

Vudentz, 2020-01-24:
Id avoid depending on the application to persist anything since that would most likely not be shared with other application or would lead to some sort of lib that would then be required to link with application which do require to restore subscriptions.

So my take on this is that it should be an option which depends on BT_SETTINGS like everything else we persist, that said it should be possible to load the subscriptions on demand so it would just check if there is anything to be sent once the application registers its callback so the application is is still required to call bt_gatt_subscribe as usual but it doesn't incur in any traffic and this way we avoid having to store the callback pointer since that could change after flashing, etc.

wdhpawa, 2020-01-27:
That sounds like an OK solution if the aim is to do all persisting within the stack. So will all service discovery results (attribute handles) eventually also be persisted within the stack? I think currently the application has to do that itself (except for the CCC handles that we are now discussing). I would just like to make sure that we are not persisting the same info twice.

PR #28038 (2020-09-04): Bluetooth: GATT: Allow application to restore subscription without resubscribing
Allow the application to add subscription without resubscribing.
This is needed for persistent bonds that can go to deep sleep and
turn of the RAM or power-cycle.
In this case the application as a GATT client must have a way to add
the subscription callbacks back to the stack before the GATT server
sends notifications.
This should preferable be able to be done before even connecting
since the notifications can arrive immediately after connecting to
the peer.
The stack cannot persist this on it's own since it must  [... cut]
commit 76119c52 (2020-09-04): Bluetooth: GATT: Allow application to add subscription.

Allow the application to add subscription without resubscribing.
This is needed for persistent bonds that can go to deep sleep and
turn of the RAM or power-cycle.
In this case the application as a GATT client must have a way to add
the subscription callbacks back to the stack before the GATT server
sends notifications.
This should preferable be able to be done before even connecting
since the notifications can arrive immediately after connecting to
the peer.
The stack cannot persist this on it's own since it must remember the
function po [... cut]
closed 2020-09-04 by carlescufi